meerk40t/meerk40t #1457: [Suggestion] m2nano - track when position is known and do an autohome if definitely not known

Sophist-UK, 2022-10-30:
I am trying to use Origin autoops rather than Home because it avoids changes of position due to minor end-stop randomness. But this means that if the K40 has just powered up or I did an abort, a Home is not done and it starts from the wrong position.

I would therefore like to propose the following:

1. A preference switch to turn it on/off (so that those who manually position the head don't have any issues).

2. If USB becomes connected when not burning or at the very beginning of a burn, or if the last burn was aborted, do an automatic Home at the beginning of the next burn.

An alternative implementation would be to have a Conditional Home autoop that you can select to be used at the beginning of a job. 

P.S. After an abort it takes a minimum of two presses of the Home button (and sometimes 3) to make the machine home.

tatarize, 2022-11-17:
We tend to avoid doing automatic homes. You can hit any of like 5 different home buttons. And home the device.

It would absolutely be worthwhile (at least for the M2 Nano), to store whether the position is well known or not. That's actually the source of `Alarm:3` in GRBL. If you abort while moving or otherwise lost the home position it requires you rehome the device to clear the alarm. Head flying should be allowed, but it would be best to track the understanding of whether the home position is known or not. Since this matters for a lot of operations, eg. if we do device level work to prevent you from leaving the bed, we don't want to enforce those rules if we don't know that the locations we have are accurate.

Sophist-UK, 2022-11-17:
This is not easy to get right IMO, because of different workflows.

My workflow is always to have real origin and MK origin the same. Mostly I burn in TL corner of the material but if not I move the design to where I want it burned. I do this because I want MK to avoid hitting the rails (and ideally warn me if it will). Other people keep the design at origin and unlock the rails and move the head to where the material is.

Here is my best guess about the circumstances where you can be uncertain about position:

* When you first start MK you have no idea where the head is. Often the K40 has just been turned on too, so the head is at origin, but that is just a guess, because I could have unlocked the rails, moved the head to clean it and then opened MK.
* After the rails are unlocked, you have no idea where the head is.
* When the K40 USB connects and a job is not running - you have no idea where the head is (because you have no idea whether the user power cycled the machine.
* After an abort you definitely have no idea where the head is.
* (Exception) When the K40 USB connects and a job is running (by which I mean that the K40 was previously connected and some instructions have already been sent) I think the assumption should be that there was a USB glitch, and that the machine processed all previous instructions and the user has unplugged and replugged the USB to get it going again, in which case the hardware and MK positions are still in lock step. 

For the first 3 you can do an automatic home, ask the user whether to home or assume the current point as origin or make an assumption about the head position (which will be right in most cases).

For option 4, the same except you cannot make an assumption about head position because it will almost certainly be wrong.

tatarize, 2022-11-17:
* Well, the biggest thing would be to flag the device as whether it knows the position or not. If you started up. Odds are good you don't know the position. So flag that as a false.
* When you home, you know where it's at. If you unlock the rail, flag that as false. You don't know what they did.
* Assume reconnects are not moved. They may have power cycled so, that's a mark against.
* After an abort, unless the head was not moving at the time you don't know where the head is.

You don't need to automatically home in these cases. You need to flag it as having an unknown location or unsure of that location, but that doesn't require homing. If you home because the rail was unlocked that's not very nice. 

You simply flag the location as fuzzy it doesn't mean it's bad or anything, just that you can't really be sure everything is where you think it is.

So the question actually is, what do you do when the location is fuzzy? I don't think the right answer is home the device. Though that would ensure that the device is no longer fuzzy. I think you actually need to refuse to respect the bed bounds. But, if you have solid coords and you know where the walls are, you shouldn't run into them. Though this should probably be done at the server level.

Yes, that information should be tracked. Yes, that information can be useful. But, I don't think homing is always right. If you want to fly your head, we have fuzzy coords. I don't need to home the device. I need to ask that maybe you work in relative coordinate systems only since I need unfuzzy coords to do work absolute coordinate systems.
There's a couple other things like absolute position moves that should maybe refuse. Relative position moves are fine. But if you have fuzzy coords you don't actually know where the absolute positions are, so maybe you shouldn't go there.

Also, if you send a job in absolute coords as opposed to relative to origin coords or relative coords it should object if your coords are fuzzy. So if you just turned the laser on and it doesn't know for sure you homed the device, it should maybe ask that you either home or mark the coords as unfuzzy.

I think those workflows cover 99% of the cases. We could also have the belt slip and can't detect that which means the coords are fuzzy and we don't know it. Unless they reset, in which case, we mark the coords fuzzy anyway. Most other devices are static and won't mess up anyway. So those devices all get flagged as unfuzzy.

Sophist-UK, 2022-11-17:
I used to do an auto Home at beginning and end of jobs, but if I want to repeat e.g. a cut because it didn't quite get through the material, I don't want to do a home which sets a new origin with some degree of randomness, but instead return to existing origin.

But now I find I do something that loses position, forget to click the home button and then it hits the end stops and I get a lot of grinding noises.

Hence this request.

As I have said, it depends on which workflow you use. If, like me, you always want Home = Origin, then I would rather MK inserts a Home when it is uncertain about position.

If your workflow is to move the head manually so that hardware and MK origin are different, then you would never want an autohome, but instead would want MK to flag definite loss of position after an abort and require the user to take action to reset it.

I cannot see other use cases - and this could easily be handled through a Device Option ("Auto-home when uncertain of position").

tatarize, 2022-11-17:
Hm. That's a pretty reasonable use case. Fuzzy home.

Seems like there's likely some better methods for that though, like run this job again without autohome. Or select some subset of the job parts and just run those.

It could probably be strictly enforced like `ALARM: 3` is in GRBL. Basically if you do something to lose registration especially aborting while running, and you are required to rehome the device to clear the alarm. 

And while it wouldn't be that hard to take fuzzy position and go ahead and give you a hybrid command with a little if statement that goes to 0,0 if the location is valid or homes the device if the location is fuzzy. But, it's not actually clear how this would appear in cutcode since those are different things.

They would only really matter for the M2 Nano. I guess it could be an an M2 Specific option. -- `Home = Goto Origin if the location is not fuzzy`. This would also mean that hitting the `Home` button would do an origin move if we don't have a fuzzy location. I suppose a couple other options would come from that. Do not permit fuzzy locations, which would require that you home before it lets you do anything, and automatically fix fuzzy locations, which would force the device to home if you do something that loses the current location. So you unlock the device and then when you lock it again, it would home the device. -- Seems weird but makes sense.

So M2Specific but lets you do some basic things with fuzzy locations. Like "goto origin rather than home with clear location", "do not permit fuzzy locations" and "autofix fuzzy location".