meerk40t/meerk40t #1462: [0.8.1] Time estimates again.

Sophist-UK, 2022-10-31:
[image]
I think that 10:02 + 0:19 should be 10:21 not 16:19.

tatarize, 2022-10-31:
There's an extras value for the raster that isn't part of travel or laser time. It's the turn around time for the data. That's what's there with the extra five minutes. And 16:19 vs. 17:17 isn't a terrible estimate.

tatarize, 2022-10-31:
Yeah, I think to get the rasters more correct, we might need to actually convert the rasters themselves into numpath code or cutcode of the actual relevant moves. Since it's guessing you'll run the entire set of pixels when really you often won't and it goes a lot faster there.

jpirnay, 2022-10-31:
Extra time will be displayed from now on

Sophist-UK, 2022-10-31:
Sorry - but suggestion about trimming above is incorrect. You would have to work out where the ends were approximately by using zero bytes as an approximate measurement and then work out what the extra sweep distance would be from the end of the first row to the beginning of the second. Could be a need to sweep further pas the end of the first or alternatively to sweep back to the start of the second.

tatarize, 2022-11-01:
Extra just tended to be things that didn't fall into travel or cut. Like if you tell the laser to wait five minutes, then draw a 3 second rect. The time is 5:03 but that's `0 seconds of travel + 3 seconds of cut = 5:03`

Sophist-UK, 2022-11-01:
I don't use those sorts of extras. Just occasionally Interrupt - in which case I make that clear. So no extras in the above example. But I would agree that extras should be included in the total and not laser or travel time even if it means that the numbers don't add up. (In the case where extras are used, we can reasonably assume that the user is an expert user and that they already know what the cause of the missing 5 minutes is.)

tatarize, 2022-11-01:
The raster accel/decel falls into the same category. It's time spent doing something that isn't moving or cutting.

Sophist-UK, 2022-11-01:
Personally I see accel/decel as being part of an active burn - if the time that the laser is **off** during sweeps is included in laser time, then so should accel/decel - that way if you don't e.g. introduce deliberate wait times the numbers add up.

tatarize, 2022-11-02:
Well, the math is that `distance * speed = time` so we should get something close to the right answer but we actually get something that apparently varies based on the stepping down speed. But, this isn't purely travel time, in fact, most of the time is spent turning around at the edges or whatever. Which wasn't part of the calculation.

Maybe the distinctions need to go away or whatever though travel time is useful there. But, we could just add extra into cut time though it makes the math such that you can't multiply distance * speed and get cut time.

Sophist-UK, 2022-11-02:
For the m2 nano we know the speed and acceleration distance (which depends on speed unless you override it).
```
d = at^2/2
s = at
a = s/t
d = st/2
t = 2d/s
```
Obviously d is in mils, s is typically in mm/s rather than mils/s, but the conversion is straight forward.

However, other boards work differently, and so the driver would need a method to calculate the sweep turnaround time.

I believe that it is possible to calculate accurately the raster time, but that to calculate it rather than estimate it might take too much CPU. But as shown above it should be possible to estimate the turnaround time with reasonable accuracy.

Sophist-UK, 2023-04-04:
0.8.2: Time estimates still off.

[image]

Also, Passes=**None** ??

Sophist-UK, 2023-04-04:
For #1 in the above image (which I still have loaded) here are the simulation estimates:

[image]

The simulation estimates are highly accurate (for this simple burn) 0:00:59 vs. actual of 0:00:58 so I have no idea where the spool history estimate of 0:01:58 has come from.

Sophist-UK, 2023-04-05:
I made a mistake and double-clicked the Go button creating two identical jobs by accident. Of course I cancelled the second job as soon as it started, but here is the job history...

[image]

As you can see, there are some weirdnesses. 

1. The number of steps is different.  # 2 is possibly missing the automatic actions at start and end (though IMO these should have been done again because this is 2 jobs not 2 passes - so I think this is a different bug).

2.  The time estimates are way different!! Weirdly, the first burn # 2 is 3x actual, and the 2nd cancelled one is 2x actual.

jpirnay, 2023-04-05:
> Passes  = None ?
Acknowldeged. Fixed. 
Need to figure out the estimate time warps still

closed 2024-10-04 by jpirnay