meerk40t/meerk40t #1502: [0.8.1] Rapid moves not working

Sophist-UK, 2022-11-19:
Slow cut of acrylic at 4mm/s. Rapid  moves > 80mil (c. 2mm) set for m2 nano, Jog Method=Default.

However moves of up to 5mm are done at slow speed. Very large jogs are done rapidly, and I have seen a 10mm jog done rapidly.

This is certainly not a major issue, but it is a little odd.

tatarize, 2022-11-20:
I think short goes at same speed. Bit longer and it switches over to rapid movements. But, I'll assume you're right and make sure to check it.

Sophist-UK, 2022-11-21:
Its a measurement thing - 10mm+ seems to be fast, 5mm- seems to be slow. Somewhere in the middle is a changeover. which is c. 3x-5x what it should be.

As I said a v. minor issue.

Sophist-UK, 2022-11-26:
The code at  https://github.com/meerk40t/meerk40t/blob/master/meerk40t/core/plotplanner.py#L142 uses the minimum jog distance as x and / or y not diagonally - so the minimum jog distance is between 1x and 1.414x the distance set.

Because rapid jog is done by the control board, consisting of a diagonal move at a fixed diagonal angle followed by an x or y move for the remaining distance, I am not sure whether this algorithm makes sense or not.

P.S. I am unclear why we use hardware jog - presumably we could emulate a jog but in a single direct straight line, simply by doing the same walk but at higher speed.

Sophist-UK, 2023-10-11:
This was reported in 0.8.1 and it is **still** a problem in 0.8.10.

I am still unclear what the measurement unit is for minimum jog distance for rapid jogs - I have assumed it is mils but there is nothing in the form or tooltips to tell you what the UoM is.

But either way, it is not working as expected. I have now set this to 5 and I am still getting slow jogs over 5 mm.

Sophist-UK, 2023-10-11:
From what I can tell, changing the distance makes no difference.

closed 2024-10-04 by jpirnay