meerk40t/meerk40t #1507: [0.81] Scene display of Use rotated element incorrect

Sophist-UK, 2022-11-22:
[Wallet concertina small - Thin 260x150](https://user-images.githubusercontent.com/3001893/203361000-ff0253fc-5d7f-44dd-b0b2-94cc1133c3e5.svg)

When I load the attached file the rightmost piece is displayed 1mm too far left/down. If you look closely the bounds are displayed correctly, but the elements themselves are not in exactly the right place:
[image]
This is, however, just a main scene issue. When I  simulate or burn it is correct:
[image]

jpirnay, 2022-11-25:
Yes, scene is incorrect. Simulation on the other side seems to recognize things properly:
[image]
Have you tried to burn this sketch and was it turning out correctly?

Sophist-UK, 2022-11-25:
On reflection this may not be a Use element issue at all.

tatarize, 2022-12-13:
It's fine in the preview of the cutcode, and it's also fine when saved out. It has the correct geometry and even gives the correct bounding boxes but clearly shows the path going way too far out around the edge there. I think the issue is that it's actually rounding the control point of the curve. That doesn't really explain why the entire edge is shown wrongly though. But, clearly it seems out of bounds but everything that gives the bounds says it's within bounds. 

[image]

It's a bug somewhere in wxPython or meerk40t for the display of the objects. The one wallet face is too low and to the left, the other is too high and to the right. It's weird.

tatarize, 2022-12-31:
This bug is in a class of bugs including one that is priority: critical. Where values are rounded and then scaled rather than scaled and then rounded. So with a scale of 2800 or so (which is fairly typical) the minimum amount of unit change that matters is 2800. Which is about 4% of an inch. This creates some notable artifacts and we're probably changing the underlying drawing engine for this stuff.

closed 2024-10-04 by jpirnay