meerk40t/meerk40t #1515: [0.8.1] Scene does not reflect image DPI

Sophist-UK, 2022-11-25:
When I change image DPI (either with right-click DPI or in image properties) the dither is not reflected in the scene until I toggle the dither checkbox twice.

Presumably and event to update scene is not being generated.

tatarize, 2022-11-27:
This is likely true, also this should happen when you resize the image too. Since strictly speaking that means the size at which that dpi is being displayed should change. So if we resize an image we should actually recalculated the simulated background image too since we might be getting details we wouldn't have had otherwise.

Sophist-UK, 2022-11-30:
Also labels are not updated with DPI - presumably due to missing events also.

jpirnay, 2022-12-10:
Being fully addressed with PR https://github.com/meerk40t/meerk40t/pull/1555

closed 2023-08-26 by tatarize