meerk40t/meerk40t #1525: [0.8.1] Raster time estimates still way off

Sophist-UK, 2022-11-28:
5 identical small images. This screen shot was 1/2 way through the second one.

[image]

Here is the history after it finished...
[image]

Sophist-UK, 2022-11-28:
Another example...
[image]
[image]

Sophist-UK, 2022-12-03:
Time estimates are `length / speed` for all element types, and it seems likely that `length` for rasters is a poor calculation at https://github.com/meerk40t/meerk40t/blob/master/meerk40t/core/cutcode/rastercut.py#L97-L102.

This calculation seems screwy. I suggest that the length calculation ignores overscan (because there are likely to be areas at the edges that are not rastered on some rows which will compensate), but (obviously) the length based on area in sq. mil needs to be divided by the step (whereas this seems to add some factor multiplied by the step). Also this seems to assume B2T or T2B raster so it will be wrong for L2R/R2L.

jpirnay, 2022-12-03:
I agree, this looks dodgy - it's just calculcating pixels and not real distances, so the calculation fails. The calculation of the expected time in the operation property panel (which is using the intrinsic method of the operation for time estimates) is probably the way to go.
[image]

[image]

jpirnay, 2022-12-07:
Not having a lot of spare time to code unfortunately, but I believe I have figured out why these estimates are so big...
[image]

Sophist-UK, 2022-12-07:
Here are two executions of the same raster, first at 450mm/s, second at 350mm/s (which should take longer)...
[image]

jpirnay, 2022-12-07:
[image]
Hopefully that's fixed, too. Top-line 300 bottom line 450 mm/s

closed 2023-02-25 by tatarize