meerk40t/meerk40t #1526: [0.8.1002] Right click execute of Image Op does tiny image in TL corner

Sophist-UK, 2022-11-28:


jpirnay, 2022-11-29:
[image]
Not sure how to recreate this?! Could you provide more details please?

Sophist-UK, 2022-11-29:
I had a design with an image (DPI 500) and some vector engraves and cuts.

Normal burn worked fine. Normal burn with only Image op enabled worked fine. But repeated attempts (including reboot of K40 and reload of MK) to execute the Image op using right-click made the image very small in TL corner and it completed extremely quickly.

Sophist-UK, 2022-11-29:
It may have been after I used the abort button. I will try to recreate it.

Sophist-UK, 2022-11-29:
Yes - it appears to happen after an abort. Load this file. Click the BGB. Abort with the BRB. Home. Right-click and execute.
[TEMP](https://user-images.githubusercontent.com/3001893/204527910-4eb4487f-f2f6-4720-bc29-938a078d7913.svg)

okto577buski, 2022-11-29:
Thanks for the useful information

closed 2024-10-04 by jpirnay