meerk40t/meerk40t #1533: [0.8.1

Sophist-UK, 2022-11-29:
Greedy optimisation currently runs through the outstanding segments in the outstanding elements and for each segment it checks the start and end distance.

My suggestion is to

tatarize, 2022-12-02:
Abort should abort, regardless where it is. In cut planning the crosshatch rasters actually get turned into a horizontal and vertical raster of the same shape. So it shouldn't do anything different than a normal raster of whatever direction.

Sophist-UK, 2022-12-03:
I think this is actually a change in how abort works because I have subsequently seen this behaviour (where it homes or even crashes into the left/top hard stops) during other operations - or indeed where it does a bunch of twitches after the proper burn stops..

I haven't had time to test whether a pause followed by an abort avoids this, but if so then the abort code might benefit from an implied pause first.

closed 2024-10-04 by jpirnay