meerk40t/meerk40t #1534: [0.8.1b2] Crosshatch directions inconsistent

Sophist-UK, 2022-11-29:
I cross-hatch rastered a file with two images. The burn was excellent, however the cross hatches were done opposite ways around i.e. H then V on one and V then H on the other.

It is a minor point, but we should try to aim for consistency of burns so that the burn is exactly repeatable. (This is why closed paths are always burned in the forward direction for example.)

tatarize, 2022-12-02:
This may be an optimization since the start point for the H then V was better positioned than the start point for the V then H. They are raster cuts and those could be optimized within the cutcode object they occupy.

Sophist-UK, 2022-12-03:
Yes - I think the two rasters are considered separate cut objects and so are optimised separately.

jpirnay, 2023-02-21:
That should / could have been addressed in tats latest raster amendments

tatarize, 2023-03-13:
It's generally improved after my work there. It's probably yes, optimizing that. Though really the crosshatch could go a bunch of different ways since it's really two different rasters just sort of scrunched together. It seems like if you want better control you should actually make two different single raster operations. Since there's not really a way to make the options possible without giving you all the different settings for the next raster too. I guess it wouldn't be impossible to stack the raster settings so that if you choose crosshatch it just creates raster1 and raster2 and editing settings lets you change either of either, but that's a bit weird.

closed 2024-10-04 by jpirnay