meerk40t/meerk40t #1535: [0.8.1b2] PNG load incorrect cf. inkscape

Sophist-UK, 2022-11-29:
This is a:

*   [] Bug
*   [] Suggestion for an enhancement

## SUMMARY

## DETAILS

## ERROR MESSAGE
```
<!-- Please copy and paste any error messages from e.g. a crash report file. -->
```

## YOUR ENVIRONMENT

*   MeerK40t version:
*   OS: 
*   Running from: <|-- e.g. Executable, package (PIP3/Pypi), downloaded source, git managed source

### SCREENSHOTS

### Zipped SVG File(s)

Sophist-UK, 2022-11-29:
Here is the same file loaded in 0.7.8:
[image]

Sophist-UK, 2022-11-29:
Same file in 0.6.24:
[image]

Sophist-UK, 2022-11-29:
So I am going to have to go back to 0.6.24 to get it to burn correctly.

jpirnay, 2022-11-30:
That's something for tatarize to look into, but from what I can gather the image viewbox is not adjusted properly?

tatarize, 2022-12-12:
Either MK or svgelements isn't properly accounting for the `preserveAspectRatio="none"` being set. That should make it so the image width and height are the actual image width and height and the aspect ratio isn't preserved. But, it loaded up and make the image a solid square image anyway even though it was specifically told not to.

tatarize, 2022-12-12:
This is corrected in commit: aea80f73ea4fc2c8ab126c00f9fa1bf386b747ba

The problem was I thought it was a good idea to call `"none"` as `None` when they mean completely different things. The first means there is no preservation of the aspect ratio. The latter means that the preservation isn't set therefore use the default which is `xMidyMid meet`.

closed 2023-02-25 by tatarize