meerk40t/meerk40t #1546: [0.8.1b2] Navigate - What to do when no elements selected

Sophist-UK, 2022-12-07:
When you have no elements selected, navigate buttons do nothing.

Would it be better for them to assume that every enabled op is selected instead?

tatarize, 2022-12-30:
They disable. If you didn't select them they aren't selected. This seems like edge condition behavior where none is needed.

Sophist-UK, 2022-12-30:
And what I am suggesting is that when no elements are selected MK instead does the chosen action on all elements. This makes sense, because this is (in most cases) what the user is expecting to happen (because they didn't explicitly choose a specific subset of elements).

tatarize, 2022-12-30:
So select everything is the same as select nothing. And you can't select nothing. It seems like having nothing selected makes enough sense. I see the logic in it, and it would just be for the nudging and dragging but It seems like turning the buttons off is sort of helpful and a bit obvious what's going on.

I'd consider it stronger if you were more of a noob and were confused by this. It's not terrible but I went to all the trouble to program the buttons to turn off and when would they need to turn off if setup like this? Never! -- I could write a trial PR for it but it seems like at best a lateral move.

Sophist-UK, 2022-12-31:
My car is the same. There are tree air con zones that you can independently turn on and off - and all off is the same as all on.

This is actually a usability thing that will cause LESS confusion for newbies who otherwise won't understand why the navigate buttons are not working.

tiger12506, 2022-12-31:
I veto this.

Disabling a button when the button cannot work in that context is established computing parlance since the introduction of GUIs.

Overloading a button to do multiple things (work on selection, and work on everything when no selection) increases cognitive load, and encourages newbies to continue being newbies because they will never understand something with complex rules of engagement.

In general, when I see "usability" improvements like this, I see users instead resort to trial-and-error to get what they actually want.

Sophist-UK, 2022-12-31:
1. You are entitled to your opinion the same as me, but who gave you the right to veto anything? 

2,. Yes - disabling the button would be a good second choice. I am not suggesting overloading any buttons to do something different. But doing nothing and forcing a user to work out why something is not working when you could just make it work for them is not IMO a great user experience. Just doing what is expected even though the user didn't do quite the right actions is actually a good thing IMO so long as there wasn't any other sensible alternative that they might reasonably have meant..

tatarize, 2023-01-06:
Vetoed.

closed 2023-01-06 by tatarize