meerk40t/meerk40t #1596: [Bug]: Raster position does not match engrave position

dorsi90, 2022-12-28:
### Summary Description

A path with black fill and blue stroke loaded into meerk40t automatically generates raster and engrave operations, however the raster position doesn't coincide with engrave.
As shown in the picture below, even though the positions math in the scene, they doesn't in the simulation and when start the job.

**This is the scene where it's match**
[image]

**This is the simulation**
[image]

Tested with Executable for windows (O.S. Windows 10) and from source with linux which i use most of the time because on windows long rasters lose steps.

This issue is not present in version 0.7.8  where on file loaded it generate only the raster operation but i can create also the engrave operation by drag the element and the position match as expected.

### Additional Details

**This is the file i used to show the issue but also happen with text**
[image]

### MeerK40t Version

0.8.1002

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Linux

jpirnay, 2022-12-30:
Steps to help narrowing the issue down:
a) Create a second copy of the path and set fill to none and stroke to red for the copy:
[image]
(properly on top of each other)
b) Set the stroke color to none for the first (original) element:
The boundaries of the object change and the shifted phenomenon appears:
[image]

tatarize, 2022-12-31:
Narrowing it down some more the graphic is put down and to the right by some amount of the stroke-width, of the invisible stroke. So when there's no stroke the pen is set to `wx.TransparentPen()` the width of this transparent pen is actually a lot fatter than the width used here So apparently the filled graphic is offsetting by a fractional amount of the pen width. This could be that transparent pen has a real width of 0 or that it's fat like the other pens and consequently the raster is closer. 

I'm running off to the store, but that's my expert analysis. I think that rather than TransparentPen we'd need our own transparent pen that either has no width or the exact same width, though that leaves us in the position that we the stroke width matters and so we set our fill for none there.

It does narrow this thing down and the main point is that we can't allow any desync between the raster cut lines and the filled lines, because this would be a critical bug. And it needs to be fixed prior to 0.8.2. So new pen and I gotta figure out if it's moving for the lacked pen. It seemed like it didn't care that much how fat the pen was but I need to run some more tests before correcting it.

tatarize, 2022-12-31:
`element* reify` fixes the problem, which likely means it's related to #1507.

There's growing evidence that the drawing routine may need to be drawn at 1:1 and not scaled with a matrix. There's some weird rounding issues here. When we have something scaled up by like 2800x like this design it basically ticks up to a new stroke width but only at like 4px and goes from basically nothing to way too thick. In theory, this design may actually work. It could just look wrong, which is a problem in and of itself.

Also, there's a problem at the right edge too when the design is coaxed into being overlapping with the red outline, it actually loses a portion of the top right leaf. Which is a problem too.

tatarize, 2023-01-01:
reopening until real fixed

dorsi90, 2023-01-15:
Hi @tatarize, with latest version that you uploaded few days ago i think the issue was solved but after the first run, that works as expected, the app want start anymore on WINDOWS 10 O.S. either with executable or from source.
I found that there is something wrong in meerk40t.log generated inside Users\NomeUtente\AppData path that i paste below. Deleting this file the app works again.

**This is the error reported from executable**
[report_from_executable.txt](https://github.com/meerk40t/meerk40t/files/10421418/report_from_exe.txt)

**This is the error reported from source**
[report_from_source.txt](https://github.com/meerk40t/meerk40t/files/10421415/report_from_source.txt)

**This is the meerk40t.log inside AppData path**
[meerk40t.log](https://github.com/meerk40t/meerk40t/files/10421413/meerk40t.log)

I hope that can help you to discover this little bug.

tatarize, 2023-01-15:
This should be fixed. It threw a big wrench in things for a bit but it's much better to scale the geometry when converting to path and then also scale the stroke-width there too. Which seems to fix this and some other problems.

If you delete the meerk40t.log file it might start recovering. But, I'll correct the info label there since it's failing the jobname for history. I think jpirnay tried to warn me of similar things, this bit of code has been a pain for a bit. Lemme fix it up really fast. That's certainly enough code to fix it.

tatarize, 2023-01-16:
Okay, fixed. Pushed a new beta.

dorsi90, 2023-01-16:
Thanks @tatarize it works as expected now and we can close this issue.
There is another little issue in the new release, not related with this.
When i load svg file the operations tree not refresh and show the '+' button on the left of the operation so as there aren't operations assigned as shown in the picture below
[image]

then just a simple click on the operation and it refresh
[image]

This is just a GUI that dosn't refresh the operations tree but the operations are assigned correctly and also showed in simulation.

jpirnay, 2023-01-16:
That seems to be a wxpython 4.2 issue - addressed in PR

PR #1611 (2023-01-01): Modify `make_path` transformed segments
Intended to fix #1596 and #1507.

This affects the drawn size of the paths so as to avoid the `round()` then `scale()` issues within wxPython. Where if we have a scale of 2800 then we round to 1, 2, 3 etc then scale by 2800 giving our values units of 2800, 5600, 8400 etc. This causes a number of very detectable artifacts. The number system used is 1/65535 of an inch because it won't show any artifacts even at the sub-mil level and can still do designs in linux the size of a football field (2^24 range).

This messes with the stroke-scaling and stroke-widt [... cut]
closed 2023-01-16 by dorsi90