meerk40t/meerk40t #1653: [Suggestion]: Accuracy

Digitdav, 2023-01-11:
### Summary Description

Need the make under preference x y scale factor enable, not sure how the undimmed, need to enter a change to accuracy, it is off on my system need to change to 1.017 scale factor.  My backlash is about 0.004” in x still working on it. I built my laser engraver using an Led laser using Lihuiyu M2-Nano controller 430mm x 560mm open frame, the only thing I had to do is invert the TTL signal (PWM).

### Current functionality

Dimmed can not enter info.

### Future functionality

Undimmed setting

### MeerK40t Version

The lates version 0.7.9

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-01-12:
Part of the reason we switched to 0.8.x was that we might actually be able to fix these issues. I think 0.7.x might be doomed in this regard. There's a rotary mode that will allow you to stretch things by the added 1.7%. It lets you say your y or x axis is a bit larger and will apply this transformation just prior to sending it to the laser.

Alternatively, we're double checking but I think it should work in 0.8.x since we have a much more robust system for the coordinate systems. Also, you can manually scale everything in console. `element* scale 1.017 1.0` would, in theory, scale everything up to be reasonably helpful there. But, full real correction of this issue was one of the predicated reasons for 0.8.x.

Digitdav, 2023-01-15:
I believe that to be user friendly it needs to be undimming. I will check out 0.8.x. 

Thank you
Dave

closed 2024-10-04 by jpirnay