meerk40t/meerk40t #1700: [Bug]: Power (ppi) delayed update

vlad1m1r, 2023-02-15:
### Summary Description

Setting the ppi for an engrave operation to less than 1000 is not updated on the next cut operation and for 5 second it engrave instead of cutting.

### MeerK40t Version

0.8.1

### MeerK40t Type

Executable

### Your Operating System

Windows

jpirnay, 2023-02-21:
Can you provide some more details about the workflow ie the steps you have taken, to help us understand the issue better?

vlad1m1r, 2023-02-21:
[image]

For example I have a file with 2 operations, an engrave one and a cutting one.
If I set the power of the engraving to 600, after it end and start the cutting operation, even if the power is set to 1000 in the cutting operation, it look like a transition from 600 to 1000 for like 5 seconds, progressively increasing the power instead of directly outputting full power.
To fix it i need to always add a cutting operation that is useless just for the power to finish increasing back to 1000.

[image]

I use the wifi device option on windows client and running lhystudios server on a raspberry pi (both 0.8.1000) but it have the same problem even if i directly connect to it with an usb cable.

tatarize, 2023-02-22:
Can you try this with the the latest beta and this will give you a `save` button on the LaserControl. Saving that .egv file and posting it here. That will let me see the actual code being sent to the laser and more quickly narrow down if this is a software issue.

There are some cases when a tube is getting old and going out that it takes a bit to heat up. Also your image isn't very conclusive. You are showing the backside of some wood but it's not obvious to me that variations in laser and wood couldn't cause that.

1) Latest Beta
2) Save File

This will at least establish the verification of the phenomenon.

vlad1m1r, 2023-03-12:
Hi, 
I made various tests and it is not the wood or the laser tube. I turned off the power after the engraving and i turned it on only after 30 seconds and it cut all the way through the wood so it is not about the laser tube needing time to "heat up".
The requested file: [k40t.zip](https://github.com/meerk40t/meerk40t/files/10950600/k40t.zip)
This was the setup this time:
[image]
Only half of the cut line was fully cut and it was like the power limit setting from the engraving was still in effect.

tatarize, 2023-03-13:
The given egv files ends with:
`...UBNTzzzzzzzzz073Lzzzzzz107SELT@NSECV2352371006000223CNLTS1EDzzzzzzzzzzzzzzzzzzzzzzzzzzzzz085FNSE-`

That end bit says set a speed `V<numbers>C` then kick into program mode NLTS1E going to the left. Turn on the laser and cut a really long line `(29 * 255) + 85` / 1000 inches long. The code sent to the laser absolutely says turn laser on. Travel left for a distance of 7.480 inches.

tatarize, 2023-03-13:
If the PPI was actually set to 500 there the file would read UaDaUaDa... and do for all 7480 units.

I still strongly suspect a hardware issue. Like when your tube's plasma turns white might have this effect. I'll ask our laser-expert about plausible diagnoses. But, I can tell you that the commands sent to the laser absolutely do reflect a cut at 100% of knob power for a distance of 7.5 inches.

tatarize, 2023-03-13:
LaserSpeed(speed=4.999805106217112, suffix_c=True, acceleration=1) cuts at 5mm/s

soylentOrange, 2023-04-11:
I'm actually having the same issue on my K40 with the stock board as well for the current beta and all of the previous versions I have used (maybe I started using Meerk40t with version 0.7...). After using a PPI setting of less then 1000 it takes some time to "flush" this setting out. Even when doing only an isolated run with say 600 PPI for an engrave. When starting the next burn operation with a setting of 1000 PPI it will take some seconds/cms before the full power seems to be effective again.
Also the board works fine otherwise and the tube is cutting perfectly fine when not having tinkered with the PPI setting. 

I always liked the idea of dynamically setting the output power on my stock K40 but I'm not using using it due to this unexpected behaviour.

tatarize, 2023-04-12:
Can you establish a situation where this delayed effect would happen and save out an .egv file so I can see what is being sent to the device. I tested this before and the egv file in no way showed anything that would result in this effect occurring. We've had enough reports that this is a thing to believe this is a thing, but I've no understanding of what would or could cause it. PPI requires that it actually turns the laser on and off (sounding a bit like sizzling bacon), I don't see that reflected in the .egv code.

soylentOrange, 2023-04-12:
Sure,

made three runs: 

1. engrave (20mm/s @ 600 ppi), burn inner (12mm/s @ 1000 ppi), burn outer (12mm/s @ 1000 ppi) executed via arm&start from the laser panel with 32% laser power and 3mm plywood. The burn operation started with the sizzling noise of reduced ppi and faded out even more towards the end: kerf_gauge_1.egv
2. engrave only, messed up completely: somehow lost track of one of the axes (I assume, this is also a known bug of the current beta): kerf_gauge_2_engrave_1000.egv
3. Only the burn opererations (12mm/s @ 1000ppi) cut nicely through the material: kerf_gauge_3_cut_1000.egv

a picture (labelled 1,2,3 and the three .egv are attachached)
[egv_files.zip](https://github.com/meerk40t/meerk40t/files/11213575/egv_files.zip)
[image]

soylentOrange, 2023-10-01:
The problems seems solved for me now with the current version (0.9.1). 
In the configuration of the machine/setup-tab/plot-planner section, I set Phase Type to static and entered 1000 for the Phase Value. 
Thanks for making meerk40t ever more useful!

tatarize, 2023-10-05:
It would have been an issue with the updates not working from earlier versions of 0.9.x. No updated settings were properly kicking in right. It was sort of a disaster and I nuked all the versions that may have had that issue.

tatarize, 2023-10-05:
Might be too old of a problem. Not sure.

closed 2024-10-04 by jpirnay