meerk40t/meerk40t #1760: [0.8.2] Delete selected groups (tree) does NOT delete groups

Sophist-UK, 2023-04-04:
As I have stated before, the easy way to delete / remove is to remove **selected** items and their children and disregard emphasised.

jpirnay, 2023-04-04:
Would be good if you could check that with the latest beta 0.8.3b2.
My test:
[image]

Hitting delete:
[image]

Sophist-UK, 2023-04-05:
I assume you mean #1756 which I will test in the next day or two.

The right-click menus also need to be made clean and consistent so that a remove / delete option works the same regardless of the type(s) of node selected.

closed 2023-08-26 by tatarize