meerk40t/meerk40t #1764: [0.8.2] Device config: Shift home functionality missing

Sophist-UK, 2023-04-05:
In 0.7.x device config has shift-home x/y values. These are missing in 0.8.x.

tatarize, 2023-04-06:
The current state of this is largely seen in #1737. While not strictly the same we are now calling the job_start() and job_finish() with the size of our cutplan object. This is needed for something like ruida and newly writing the actual sizes to their files before sending them. This is because Ruida actually runs a check on the machine to see whether that will crash the head and object, and newly needs it because it builds in a "Frame Job" into sent project to allow you to frame a job.

Basically shift home x/y would be built into the job size check, and placements. To get the regular coordinate systems to work we have some code to map a placement to a resulting area of the device coordinate system. We can find the matrix which maps four points directly on to four other points. And solving for these within the jobs. This allows us to map these points between scene space and device space. We could even set a sort of rotated placement and the elements would adjust accordingly.

* For now, we've added the jobs coordinates (which broke Beta1/2 of 0.8.3). This sends drivers the job information on a per-job basis. Most jobs now provide this information if it has this information.
* Next we're going to code in placements, in a very simply manner. Rather than using the scene-to-device matrix for everything we will figure out the placement (4 corner points) where we want the project to be located in the laser. Initially this should be the bounding box multiplied by the scene-to-device matrix.
* This seems like a bit of weird math to arrive back where we started. But, we could then pre-define other placements. Multiple placements, and placements that are shifted from home by some x/y values.

This should allow us some access to other coordinate systems. So rather than absolute, which is what we get multiplying our bounding corners by the scene-to-device matrix to get our 1 placement. We can also query the laser's believed position and map the correct sized rectangle relative to that point. Which is current-position coordinate system. And finally we could do this based on some predefined values, either bounding boxes offset by a series of points or corners which are exactly predefined. Which would allow you to place the same job in multiple places but also with multiple different rotations, or scales if you wanted that for some strange reason (usually placements would be 1:1 between physical units in scene and machine space).

The basic plan is the bullet points. We can query for a placement, add in an offset when we do that and we'd still be able to make sure the laser doesn't return any warning about the offset xy slamming into the wall. That was a bit of the problem with the shift home x/y. Not just that it was only for the lihuiyu device, but that it generally worked by forgetting where it was located.

Here we should be able to define a step-repeat grid, and view that grid since it will consist of a bunch of predefined placements located at regular intervals. Though the command would only set a series of placements and loop through those, you could probably set them anywhere, or in whatever order you need. Stagger them or rotate them. But, stuff like that where you have a bunch of jig information and placements is decided a lot more work and making it look nice etc. But, we can do the easy create an absolute coordinate placement to use, and do that but give it an offset by x/y.

---

The second part of this is to make the shift home look right and we have some work to do with views and the different viewmaps. It turns out when you generalize the math you can map the scene-to-device space perfectly so that 0,0 is always just the upper-left corner for the program. We then need a bit of a modified view, for the user-units. Which basically would be used to map the user-units to the scene units. This is because we have some issues where the user uses the Move-To panel and send it to 0,0 and it will always go to the upper-left hand corner even if they see 0,0 in the upper-right hand corner because they have their bed flipped.

The shift-home functionality would be done in this user view. Where you could regardless of where your laser actually homes or how the positions it actually uses go ahead and mark 0,0 in any corner you want or in the center of the laser bed. Then issuing an goto 0,0 would check the user-coordinate system and figure out where that is in scene and device coordinates and move the laser to that position.

Shifting all your jobs you send to the laser over by 5cm would be covered by first bit. But, you really sort of want your bed to have the 0,0 position located 5cm into the bed when you do that, and that latter bit requires this other change over. There was a very long discord discussion with Tiger about this, and the math works out really well. Though converting most of the physical positions you issue to the program to query a user view and then solve where that goes in the scene and in the device is a bigger change that may have growing pains. So gradually starting with adding placements and forcing everybody to use the scene view with the home in the upper-left hand corner is a solid phase 1.

tatarize, 2023-04-22:
Placements are added which does a sort of shift home functionality. There's some discussion on what this should end up being and whether this is the right answer but so far it seems to work fairly well.

closed 2024-10-04 by jpirnay