meerk40t/meerk40t #1766: [0.8.2] Preferences/GUI/Appearance/Icon size - toolbar icon weirdness

Sophist-UK, 2023-04-05:
1. When you change icon-size away from default e.g. to big, the toolbar icons are missing.

[image]

2. Because it says "Mini icon in tree" before "icon size" it suggests that the icon-size value applies only to the tree. I would suggest that the icon-size setting is moved to before the "Mini icon in tree" setting.

jpirnay, 2023-04-05:
> Because it says "Mini icon in tree" before "icon size" it suggests that the icon-size value applies only to the tree. I would suggest that the icon-size setting is moved to before the "Mini icon in tree" setting.
That has been addressed in PR https://github.com/meerk40t/meerk40t/pull/1770

Sophist-UK, 2023-04-05:
> The ribbon toolbar collapses like shown if it has not enough vertical space to display icons, buttons and labels (or it thinks so). Please resize the pane area to make it slightly larger and they should reappear. Afterwards you can make it again a bit smaller

Thanks for the steer - I will give it a try, however IMO MK should enlarge the ribbon / pane height if necessary to display the icons.

closed 2024-10-04 by jpirnay