meerk40t/meerk40t #1782: [0.8.3 Beta 3] Various burn issues

Sophist-UK, 2023-04-08:
Since trying this beta I have had various examples of burns going wrong.

1. Y-axis mis-positioning at slow 2mm/s vector speed
2. Image x-axis mis-positioning (at 300mm/s - I can normally;y achieve 550mm/s without losing position).

I have no idea whether this is due to changes in code or a hardware glitch, but I thought I would report it in case it is significant.

tatarize, 2023-04-09:
We've had a couple reports or mispositioning but we can't seem to duplicate one to track it down. I guess it could be specifically in the lihuiyu driver maybe something is broken in there and causing mispositioning there.  Specifically like this, reported in 0.8.2 regular version.

[image]

Sophist-UK, 2023-04-09:
Yes - that is **exactly** what I had in issue 1. Intermittent issues like this are always difficult to track down, but given it's importance and the damage to MK's reputation for reliability, it probably needs to be tracked down.

tatarize, 2023-04-09:
I'm running the `lhyinterpreter` and finding a 2x4 desync with the actual code. But, that's certainly more than 2x4 off. I guess I'll track down the error and see.

lhyinterpreter reads all the packets you send to the device and interpreter them to figure out where the laser position should be. If at the end a goto 0,0 results in a non-zero position, it means either the interpreter or the driver is broken in some way.

So the guess is that maybe this is real and we're somehow getting varying amounts of error, and in most cases it would be undetectable but in others it might be large enough to destroy pieces. 

--- I'll see what's causing the desync I can detect on the file I was sent (though it's not the heart that showed that issue in the above image).  It's the only thing I have. It's that, or fiddle with placements (map your job anywhere on the laser), or mess with ezcad2 file parsing some more.

tatarize, 2023-04-09:
Tried a smaller version and manually counted. Was a bug in the lhyinterpreter. Back to square one...

tatarize, 2023-04-09:
If you can get a consistent example it might help. But, I can't seem to figure out why it should mess up. I'm not entirely sure it's not a series of hardware quirks. We also have like 2 reports of rasters and then cutting being at lower power, but the `.egv` files say it was fully powered.

Sophist-UK, 2023-04-09:
It might certainly be an unexplained quirk of the m2 nano microcode that is being triggered in some circumstances. I have read of your / Jacob's discovery of some of these, so this is entirely possible. Or it's an obscure logic bug in the code.

The raster issue is less common IME. The loss of vector position seems more common and more important.

Sophist-UK, 2023-04-09:
Have you tried comparing the Lhycode produced by e.g. 0.8.1 and 0.8.2 to see what the differences might be?

tatarize, 2023-04-10:
There are some specially designed codes that can cause it to lose registration. At one point I tried `IBzzzzzS2P` as a code where that should move to the right and then stop without the stepper motors locking which caused it to lose registration. There's also some chance that the `@` connection stuff is a problem. Between 0.8.1beta2 (when I could save) and the latest there seems to be some rounding changes, or maybe slight image changes in how it's converted. And the order of the <Direction> and `D` (laser on). Changed order. It used to say `DB` and now will consistently say `BD` but this is because you could have turned off the laser if you were performing a raster. 

B|qDbUaDhUaDhUaDhUaDhUaDgUaDhUaDfUaDbUaDaU055DhUaDgUaDhUaDbU209DcUaDhUaDhUaDeU065
B|qDcUaDhUaDgUaDhUaDhUaDhUaDhUaDfUaDaUaDaU054DaUaDhUaDgUaDhUaDaU208DaUaDcUaDhUaDhUaDdU064

Some sections changes look like this. Where it mostly matches, but the section `U055DhUaDg` vs `U054DaUaDh` looks quite strange. Off by one is rounding but the difference between Dh and Da is 7 which doesn't seem like a rounding issue. At the end we get Db vs Dh which should about even that back up. But, I don't see why that would be rounding error.  It could be resampling error if the images were treated differently. It was exact for a long time then stopped being exact, I am guessing because it hit a scanline difference. 

I suppose I could switch the D and <direction> flags. order back for most cases but the order in which you change the stepper motor's direction flag and the laser on/off state really shouldn't matter.

And I guess I could chase down exactly what resulted raster image slight change.  In theory if the old code never did this and the new code does... but it's seemingly sort of random the best solution would be to restore the old ordering for the code anyway. For example we used to use a lot more `@` commands but they turned out to be able to sometimes cause a runaway.

Since the changes result in the same count of position when you add up all the commands any desync would have to be hardware. And you could be that the ordering makes it sometimes more likely. Or maybe beta 2 isn't old enough. I'll see if I can hack an output routine to 0.8.1000 and look for other changes.

tatarize, 2023-04-10:
Swaps the Up/Down flag and the Direction flag:
https://github.com/meerk40t/meerk40t/commit/a5f877459b8fb01f5a961ed3106bbb5ff8b66059
https://github.com/meerk40t/meerk40t/commit/a7ce1cf0c7fd245104efb2b7df0757aea75cf9cd

Corrects rare cases where the step amount set by G002 is too great and needs to step backwards or we need to do a zig. Where we raster to the right, move towards top, then keep going right rather than reversing directions (native rastering won't do this but lightburn could).
https://github.com/meerk40t/meerk40t/commit/ce212af9319530ee48fba0d8ff86dfc8d134e40c

These are actually both somewhat too new. Since really both of these changes are entirely 0.8.2000 through 0.8.2003 which puts them after then 0.8.2 full version which apparently had this mystery defect.

Some are in the right time period but don't seem like they could do it.

* There's some changes to get move-at-speed override working. This generally is not used but is for people using a rotary and sometimes people with way too lose belts doing very small train pieces for some reason. 
* Fixing the challenge codes.

I think the difference is probably the best way forward so I'll track down the difference in rasters there. I would have thought some of it is rounding but surely not all of it.

tatarize, 2023-04-11:
Fr4nklyns code managed to reproduce it with those same circles in simulator so I should be able to figure it out.

[image]

Though the PR I thought would do it, doesn't seem to do it. The change in code happen there but the theory doesn't seem correct since I can produce the flawed circle in main but not circa that code. I guess I have more bisection in my future.

But, being able to reproduce the problem is great.

tatarize, 2023-04-11:
The geometric bounds are used to create the rectangle and the painted bounds are used to actually draw things:

https://github.com/meerk40t/meerk40t/commit/6c34a1d0da7ddb7c5be37af1a249e11b77f598c8

Debug code wandered into the code base and caused a painted/non-painted desync for sizing. 

I had noticed it before before but it wasn't responsible for the weird other error we were addressing at the time.

---

#1791 fixes this.

tatarize, 2023-04-11:
I tried a bunch of different things with the commit I thought it could be but it was off by pixels at a time. And the example code didn't fail.

commit f13a4146 (2023-04-11): Fixes #1782, size determined by geometric rather than painted bounds for raster image.

Fixes #1782, size determined by geometric rather than painted bounds for raster image.
closed 2023-04-11 by jpirnay