meerk40t/meerk40t #1783: [0.8.2] Config/Display options

Sophist-UK, 2023-04-08:
Inserted `{dpi}` into custom Raster op and Image element tree descriptions.

For some reason DPI is shown as an integer in Image elements but as a decimal with one digit after the decimal point for Raster op. (Also integer for Image op when I mistakenly added it there.)

Also unclear why these are part of Device Config rather than global preferences.

jpirnay, 2023-04-11:
> Also unclear why these are part of Device Config rather than global preferences.

As different devices have different focus, so speed is much less relevant for a Galvo laser than for a CO2 or Diode laser

Sophist-UK, 2023-04-11:
Yup - that makes sense. Most people will only have one laser, and most people will not fiddle with these, and those that do will change them infrequently, so having them by device is not gonna be an issue.

closed 2024-10-04 by jpirnay