meerk40t/meerk40t #1790: [0.8.2] Group without inkscape label inherits from parent

Sophist-UK, 2023-04-09:
I am not sure if this is intentional or not. Should it use the id instead?

#### Meerk40t
[image]

#### Inkscape
[image]

jpirnay, 2023-04-11:
Are these 'regular' groups or virtual copies via 'use'?

Sophist-UK, 2023-04-11:
As you can see from the Inkscape screen shot, they are a mixture of real groups and clones, mostly real groups. But I didn't set the Inkscape label on them, so they are inheriting the parent label rather than saying e.g. "Group g5815 (2 children, 2 total)" or "Group use6217 (2 children, 2 total)".

But whilst we are talking about clones, it might actually be useful when MK converts a clone/use reference to a copy of the original to show this by prefixing the label with e.g. a "*" or dagger or similar.

jpirnay, 2023-04-12:
I can confirm the issue but this requires a deeper change of the underlying logic as it involves amendments to [SVGElements](https://github.com/meerk40t/svgelements) which has a release cycle of its own.

tatarize, 2023-04-12:
It actually shouldn't. It's not really that noted but as a subitem. The values also have "attributes" within the values and this is the dictionary of the tags that existed *on* that specific tag and *only* on that tag. So while the values are inherited from their parents if you check the values["attributes"] dictionary for the label in question you'll find it without the children being affected.

closed 2023-04-13 by tatarize