meerk40t/meerk40t #1798: [Bug]: The y-dimension does not match the actual size in the SVG file

ArtemShalamov, 2023-04-13:
### Summary Description

The y-axis size does not match the actual size in the SVG file. Although the size on the x-axis corresponds to the specified. 

### Additional Details

The y-axis size does not match the actual size in the SVG file. Although the size on the x-axis corresponds to the specified. The size of the part in the file is 200 x 200 mm, but MeerK40t perceives this size as 200 x 195. A screenshot where this can be seen is attached. The SVG file is also attached. The file was created in Inkscape.

[image]
[image]

### MeerK40t Version

v0.8.2000

### MeerK40t Type

Executable

### Your Operating System

Mac (M1/Bigsur)

tatarize, 2023-04-16:
It's clearly the case that that object has those dimensions and the selection widget is giving the wrong values. Thankfully it still has the geometry right, so it would still correctly run but the selection is somehow flawed. It might have something to do with the circles. But, yeah, that's clearly broken.

tatarize, 2023-04-16:
Rotating it slightly resets the box and calculates it again, seems like that resets it. Like it might be a miscalculations that gets cached.

ArtemShalamov, 2023-04-16:
@dorsi90 After changing units from pixels to millimeters in Inkscape, MeerK40 still doesn't display the file correctly. The file with the changed units is attached below.

[image]

ArtemShalamov, 2023-04-16:
> It's clearly the case that that object has those dimensions and the selection widget is giving the wrong values. Thankfully it still has the geometry right, so it would still correctly run but the [...]

When opening a file, if its dimensions are specified incorrectly in the program, then incorrect dimensions are sent to the machine as well. That is, if the dimensions are shown as 200 by 195 when opening, then the machine will cut it at 200 by 195. After a slight rotation of the file, the dimensions become correct, which is definitely good news

tatarize, 2023-04-16:
Typing `element* rotate -1` will cause it to fix, but then `element* rotate 1` will break it again. That shouldn't be stuck cache information. It might actually be giving the wrong bounding box.

tatarize, 2023-04-16:
[image]

Deleted everything else and just kept the main circle still got it wrong. I guess I'll check with svgelements and see if the bug is there and just miscalculating this otherwise.

dorsi90, 2023-04-16:
@tatarize, @ArtemShalamov the main circle is the problem. if you try to delete it in inkscape and create and new one it works as expected. I think there is someting wrong in the path

tatarize, 2023-04-16:
It actually tracks down to a numerical instability in the calculations of the y-extreme for the cubic bezier curve. When multiplied by the matrix you get the bezier curve:

CubicBezier(
start=Point(177947.120054983418,503068.324846403091), 
control1=Point(229690.656036858971,519880.818837954546), 
control2=Point(285453.184657795762,519880.818837954779),
end=Point(337196.720639534935,503068.32484640379))

The relevant y values are:
a0 = 503068.324846403091
a1 = 519880.818837954546
a2 = 519880.818837954779
a3 = 503068.32484640379

This creates a denom value (a0 - 3 * a1 + 3 * a2 - a3) that equals: 3.4924596548080444e-10

This is exceptionally close to zero but not so close to zero that it *counts* as zero (which is 1e-12), and uses some other code based on setting values to 0 in the quadratic formula.

Our r1 becomes -1.3645833333333333, and our r2 becomes -96278815810543.95 neither of these fall into the required domain of being between 0 and 1. So it says that there are no inflection points within the curve and is decided entirely monotonic (keep in mind this is the entire curve at the bottom of the circle which absolutely has an inflection point and curves all over the damn place). 

We therefore use the local extremes at 0 and 1 which are both ~503068 so that value is said to be the right answer. Because this is clearly a straight line. So it draws a straight line. 

[image]

The range of the numerical instability is actually significantly higher than 1e-12. Specifically because we do a square-root in there. And it needs to call a lot more numbers that are decided very close to zero to be zero. Probably well up to 1e-8 or so.

tatarize, 2023-04-16:
https://github.com/meerk40t/meerk40t/commit/390ce9641beca62f65811585cee453edcbc6c754

This push to main fixes it.

tatarize, 2023-04-16:
This is decided going to be quite rare. It's that the curve is very very close to exactly perfectly mirrored without being so exact to trigger the fallback deciding that our denom value is going to be zero. Had it been a bit more perfect it would have done the correct fallback, and had it been a bit worse it would have been correct with the calculations.

Dividing numbers very close to zero makes floating point numbers do weird things.

joerlane, 2023-04-17:
Fixed by tatarize. There's a new beta release that fixes this issue. Signed mac builds are ready for download.

tatarize, 2023-04-17:
This bug is so exceptionally rare that if you breathed on the circle wrong it would correct it. Which is why we had so many (I resaved this and it fixed) issues. Generally because rounding the number slightly could change the error to the usable range or to the fallback range.

closed 2023-04-17 by ArtemShalamov