meerk40t/meerk40t #1812: [0.8.3] Very first burn was broken - back to 0.8.2

Sophist-UK, 2023-04-20:
It burned this then went to right stop and ground the steppers.
[image]
As you can see the stitch holes are broken, and as you can see by the rebirth on 0.8.2 they were in completely the wrong scale too:
[image]

P.S. You need a better real life testing regime before publishing a full release.

tatarize, 2023-04-20:
Can you email a file that contains this problem.

We searched for #1782 for a while and it should have been the only real thing to affect the burns there.

tatarize, 2023-04-20:
Or a saved out copy of the file as an `.egv` between 0.8.3 and 0.8.2. I believe these should be roughly unchanged in what they send. 

* Driver: Added minor hooks for rotary.
* Driver: Added rare edges case of raster_step_float is None set it to 1.
* Device: Added outline values for jobs. This currently does nothing for the Lihuiyu driver.
* Device: Converted most self.settings() call to choices.
* Device: Added an incoming choices value to the device creation for "friendly name" device manager adding.
* Controller: Avoid crashing with serial==None, but enabled. 

That's it, all the changes between 0.8.3 and 0.8.2 for the driver.

tatarize, 2023-04-20:
I tested `wallet-test.svg` that had your drill dots on it, and the 8.2 and 8.3 files saved identical except for the section where they list the version number.

tatarize, 2023-04-20:
Tiger has a theory that maybe the newer ch341 driver had an issue and a disconnect was improperly recovered.

Seems like a solid enough idea.

Sophist-UK, 2023-04-20:
> Can you email a file that contains this problem [?]

Yes - but the file burned perfectly on 0.8.2 but not on 0.8.3. 

In both cases I ran the executable, loaded the file and clicked the Go button. In neither case was there a pause where the K40 emptied its burn buffer and then waited and some several 10x secs later MK recovered the USB and carried on but some data was missed. So I do not believe that there was a USB issue.

My experience is that 0.8.2 is pretty darn reliable, but 0.8.3b3 and 0.8.3 give problems every time I have used them.

P.S. My experience with 0.7 and 0.8 up to 0.8.2 is that USB hangs, and 9x out of 10 when I unplug and replug it carries on just fine and 1x out of 10 it stays hung - 0x out of 10 does it go wrong like this. I have not used 0.8.3 (b3?) enough to have experienced a USB hang.

tatarize, 2023-04-20:
You run enough projects and know the software well enough that the assumption is you're right. I just don't really know how it's doing this. There's no difference in the egv of the test file I tried between 0.8.2 and 0.8.3 and nothing in there changes. So I need a test file that shows the difference, or I'm currently looking into tigers guess that it was a quick disconnect/reconnect that somehow lost the packet. Like it did a move, then hit a small issue, and lost the packet rather than sent it.

If the recovery from a disconnect is a bit subpar or could drop a packet and misrecover and that seems like it could be a thing. 

That's currently where I am, unless you have a files that show this is consistently happening here.

My email's still `tatarize` at my gmail

Sophist-UK, 2023-04-20:
I sent you the file by email 3 hours ago.

I have no idea why this is happening, I just know that 0.8.2 works reliably and 0.8.3b3 and 0.8.3 have failed badly on each of the 4 times I have tried to use it.

With multiple files, 3x on 0.8.3b3 and 1x on 0.8.3, Of the 4 burns I have attempted, all 4 have been bad fails and 3 of the 4 were issues with vector burns and did **not** involve an issue rastering.

I use MK in production, and if I run a new version and get a failure immediately, I switch back to an earlier reliable one and don't waste more material on versions of the software which fails and ruins material.

I have no idea what the cause is, I just know that it is happening.

I guess it is possible that this is a USB issue, but when I get it literally every time I used 0.8.3b3 or 0.8.3 and not getting them on 0.8.2, it is still a bug in these later versions. However, I don't think the scaling issue is due to lost USB packets - as I said, if this was happening then the m2 nano would lose position completely, but what we are seeing is a scaling issue but **without** losing position completely.

Sophist-UK, 2023-04-20:
I will try to post screen shots from my production machine of my device settings tomorrow. Perhaps I have tweaked jog settings which are causing this with my machine,

tatarize, 2023-04-20:
[image]

I saved the code in 0.8.2000 and 0.8.3001 and there was no difference. The only thing different in the EGV was the version number.

Can you save a copy of the .egv files for both versions and see if you have any differences in the output code. My best theory is still tiger's idea that doesn't cope with disconnect and reconnect. But, this would have been true before 0.8.2000 too.

tatarize, 2023-04-21:
* Saved out egv might help. Comparing 0.8.2 and 0.8.3
* Try again you could on cardstock... (does it always happen or fluke)
* Do you run libusb or windll for your machine? Might be a recovery issue, the ch341 stuff was changed a few months back.

I'm leaning towards it being a rare 207 error code issue, sometimes 207 packets respond wrong and the packet either was or was not processed and gets double-process or dropped. It's rare but a known issue. Usually it needs resending, but sometimes the board actually processed it and sent a 207 and sometimes it didn't process it and sent a 207 so it's ambiguous. 

If one happened to hit you it could explain the stuff. I've needed to add 207 warnings to the usb log for a while.

tatarize, 2023-04-21:
The determination is that the EGV isn't changed. So CH341 Checks was launched. @jpirnay did considerable testing and says 0.7.9 worked flawlessly. Whereas 0.8.3 failed quickly in nearly all his test cases, even at lower speeds. Whereas 0.7.9 worked fine even at higher speeds.

[image]

And more.
[image]

He tried running it slow, and running 0.8.3 code in 0.7.9 and 0.7.9 code in 0.8.3. His conclusion was sending egv data in 0.8.3 was highly problematic and he would get some errors:

[image]

Which may actually be because of a change in the way status is checked on the device. But, you certainly have a convert to the something is wrong conclusion.

---

Based on this we'll try some additional testing with #1822 where in addition to USB, Networked, Mock, you also get Legacy. Which is the exact code that 0.7.9 uses. This should maybe help narrow things down.

Sophist-UK, 2023-04-22:
1. I have LibUSB installed. So **not** an issue with CH341 drivers.
```
    Using LibUSB to connect.
    Attempting connection to USB.
    
    
    Finding devices.
    1 device(s) were found.
    K40 device detected:
    DEVICE ID 1a86:5512 on Bus 000 Address 001 =================
    	 bLength                :   0x12 (18 bytes)
    	 bDescriptorType        :    0x1 Device
    	 bcdUSB                 :  0x110 USB 1.1
    	 bDeviceClass           :   0xff Vendor-specific
    	 bDeviceSubClass        :    0x0
    	 bDeviceProtocol        :    0x2
    	 bMaxPacketSize0        :    0x8 (8 bytes)
    	 idVendor               : 0x1a86
    	 idProduct              : 0x5512
    	 bcdDevice              :  0x304 Device 3.04
    	 iManufacturer          :    0x0 
    	 iProduct               :    0x0 
    	 iSerialNumber          :    0x0 
    	 bNumConfigurations     :    0x1
    	  CONFIGURATION 1: 100 mA ==================================
    	   bLength              :    0x9 (9 bytes)
    	   bDescriptorType      :    0x2 Configuration
    	   wTotalLength         :   0x27 (39 bytes)
    	   bNumInterfaces       :    0x1
    	   bConfigurationValue  :    0x1
    	   iConfiguration       :    0x0 
    	   bmAttributes         :   0x80 Bus Powered
    	   bMaxPower            :   0x32 (100 mA)
    	    INTERFACE 0: Vendor Specific ===========================
    	     bLength            :    0x9 (9 bytes)
    	     bDescriptorType    :    0x4 Interface
    	     bInterfaceNumber   :    0x0
    	     bAlternateSetting  :    0x0
    	     bNumEndpoints      :    0x3
    	     bInterfaceClass    :   0xff Vendor Specific
    	     bInterfaceSubClass :    0x1
    	     bInterfaceProtocol :    0x2
    	     iInterface         :    0x0 
    	      ENDPOINT 0x82: Bulk IN ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x82 IN
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x2: Bulk OUT ===============================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :    0x2 OUT
    	       bmAttributes     :    0x2 Bulk
    	       wMaxPacketSize   :   0x20 (32 bytes)
    	       bInterval        :    0x0
    	      ENDPOINT 0x81: Interrupt IN ==========================
    	       bLength          :    0x7 (7 bytes)
    	       bDescriptorType  :    0x5 Endpoint
    	       bEndpointAddress :   0x81 IN
    	       bmAttributes     :    0x3 Interrupt
    	       wMaxPacketSize   :    0x8 (8 bytes)
    	       bInterval        :    0x1
    Config Set
    Config Set: Success
    Getting Active Config
    Active Config: Success.
    Kernel detach: Not Implemented.
    Attempting to claim interface.
    Interface claim: Success
    USB Connected.
    Sending CH341 mode change to EPP1.9.
    CH341 mode change to EPP1.9: Success.
    Device Connected.
    
    CH341 Chip Version: 50
    Driver Detected: LibUsb
    Device Connected.
```
2. I don't have time to run a test with 0.8.3 now, but I will later and post any errors.
3. The latest pics show 2 of the 4 issues I have experienced, but these are **NOT** the issues reported here so there are still 2 issues that you haven't reproduced or diagnosed. These cannot IMO be explained by poor error handling because they have **temporary** loss of position and / or scaling neither of which can be explained by a random loss of packets between MK and m2.

Here are my device config settings:
[image]
[image]

Sophist-UK, 2023-04-22:
> But, you certainly have a convert to the something is wrong conclusion.

It is a pity that you are so sceptical about my bug reports because I have AFAIK a 100% record of bug reports being valid. And this is not the first time that you have been dismissive of my bug reports (or indeed have told others that my previous coding efforts were a waste of (your) time).

tatarize, 2023-04-22:
It's simply that hardware issue was a pretty likely answer. There was no software changes between the cited versions. It's sending the same machine code both times. So the hardware did that. And usually hardware issues are usually hardware. 

```
Tat — 04/20/2023 7:27 PM
Sophist is assumed to be correct.
```
```
Tat — 04/20/2023 7:28 PM
If the EGV file isn't different, then something is weird that isn't EGV.
[7:28 PM]
Which means it lost a packet.
```
```
Tat — 04/20/2023 7:28 PM
Did we make packet loss easier? We very well might have.
Yes.
The one test I did. Rastering and then unplugging the USB. Then sending it to 0,0 hit the wall.
That implies our recovery from a noisy channel is a problem.
```
```
Tat — 04/20/2023 7:08 PM
Well, Sophist is usually right, so there's probably still a problem there. And it's probably just harder to figure out.
And tiger's theory is my best hypothesis now.
```

Your track record for bug reports and feature suggestions is exemplary. The assumption is that your reports are absolutely correct. But, without differing code being sent to the laser it means tracking what's left to be wrong is notably a lot harder to figure out.

I went through a number of hardware tests, jpirnay did more extensive hardware tests, all because you said there was a critical issue here and you're extremely likely to be right, even if I don't understand how such a thing would currently be possible. Transmission, which was Tiger's explanation, did a lot to explain this. --- However, there's was still an open question whether the packet loss was dumb luck or something I made worse.

The `legacy-lihuiyu-interface` branch patched in the 0.7.9 code into 0.8.3 and jpirnay's tests say it's rock solid while 0.8.3 main code is a nightmare.

Sophist-UK, 2023-04-22:
As I explained from the start, this is an issue that does not occur with 0.8.2 and occurs with all later versions. Logic dictates, therefore, that this cannot simply be a hardware issue (or at least not alone) because if it was then it would occur with 0,8,2 as well and not disappear when I switched back to 0.8.2.

So it is either a combination of hardware issue triggering something that changed on 0.8.2, or more simply just an issue with 0.,8.2 and possibly my device config settings. And Occam's razor suggests that in such circumstances, it is likely to be the simplest (i.e. least number of broken parts) of the two. But either way, definitely something that has changed even if you think nothing has.

All the pics I have seen from your testing (both here and direct from Jens) show permanent shifts. That was what I saw and reported with 0.8.3b3 (though in my case they were vertical rather than horizontal) and didn't involve a raster, and the fix you did (based on shifts reported by others) was a AFAIK a raster rounding issue. The pic you posted above shows a horizontal shift, and it appears that this is not due specifically to a raster but rather due to packet loss, so this appears to be a second bug.

However, as I have previously pointed out, when you look at the very first photo and see that some paths are burned out of position but that change of position doesn't persist and it goes back to a regular shape (even if the scale is out). Since the m2 nano is a completely dumb controller that does NOT keep track of position, then it would seem that this cannot be a lost packet issue because if it was then it would lose position permanently and persistently as all the pics of replicated shifts show. So, logic dictates that this is a third, yet to be diagnosed problem.

P.S. I should also mention that Abort is also misbehaving in 0.8 - rather than stopping immediately and clearing the already sent packets (and ideally noting that it has lost position and doing a home automatically when you try to burn again), quite often it results in a lot of grinding noises as it hits various end-stops. Whether this is related to the packet loss bug that you have identified I don't know.

P.P.S. To put this in perspective: 

1. Bugs in code are normal - unless, that is, you are doing that provable coding stuff which is not used very often **even for** stuff which is safety critical to human life because it is extremely time consuming and slow to achieve. 

    When you consider that development of 0.6 stopped only in September 2021, the evolution of MK with 0.7 and 0.8 has been **incredibly** rapid. And despite that pace of evolution, my opinion is that on the whole the architecture, code quality and use of objects in MK is pretty darn good (which is almost entirely a credit to your coding skills and is why that evolution has been able to be so rapid), which is why it has so much functionality, performs so well and is generally remarkably reliable.

2. I continue to be happy to try new versions and report bugs in order to assist with ironing them out. 

3. However, as the user base grows the need to keep the quality of releases high grows with it and thus so does the need for rigour in both code and processes. and there are well understood ways to make Python code even more reliable in these sorts of systems i.e. improved unit tests (especially the lower-level code), better exception handling (i.e. using MK specific exceptions and NOT reusing standard ones), better logging, strict typing (rather than code which in places allows, excuses, ignores and fails even to log incorrect parameter types), a more formal and rigorous live-testing using test files which are specifically designed to stress different functionality, and code metrics (to track quality including both coding style and unit test coverage). My advice - if you care to listen to it - is for you and Jens to spend some significant effort focusing on these sorts of code quality improvements.

tatarize, 2023-04-28:
It's a big of a combination of effects, but yes. Waiting on other folks to give the go-ahead for the next version based on how seriously critical this bug is. Not sure if I might have broken grbl or fixed it better currently was trying to get through the pretty low bar for new version:

- [ ] Lihuiyu: Vector.
- [ ] Lihuiyu: Raster.
- [ ] Lihuiyu: Pause + Resume
- [ ] Lihuiyu: Estop
- [ ] GRBL: Vector.
- [ ] GRBL: Raster.
- [ ] GRBL: Pause + Resume
- [ ] GRBL: Estop
- [ ] Galvo: Vector.
- [ ] Galvo: Raster.
- [ ] Galvo: Pause + Resume
- [ ] Galvo: Estop

But it seems like waiting at all is a problem since there's a critical flaw in the current code and I'm not sure everybody would opt for the beta.

Sophist-UK, 2023-04-29:
I can't speak for anyone else, but I am not going to test any betas that don't have this fixed.

TBH I am surprised that there is any debate about priorities between fixing a critical bug in m2 nano code which is the predominant controller used by the vast majority of MK users and delivering new functionality for controllers that are a very minor percentage of existing users.

As a long-time IT professional, I always believed that there are two types of IT character. Those who like doing new stuff and innovating e.g. project managers/developers, and those who like doing repetitive stuff e.g. operations/maintenance. I was a project manager and developer, so I know which camp I fall into, and I think the same is true for you. So I can understand from your personal perspective how delivering the new stuff has a high priority - however from a Project perspective, maintaining the MK reputation for reliability and keeping the existing user base happy should clearly be a priority. In other words, cognitive dissonance.

But I cannot stress too much the damage that a full release with a bug like this has on user confidence. The longer this full release is out there, the more users will have this issue - and many will think that it is something hardware related or user error. New users may just give up and never come back. It is highly unlikely that they will come here and see that there is a bug, but if they do and they don't see that it is being treated with the highest possible priority, they will almost certainly go elsewhere.

**IMO there should be no greater priority than getting MK back in a reliable state for the majority of existing users, however great your wish to complete new functionality.**

As an aside, you might want to create three pieces of start-up functionality in MK. Firstly, a new version `What's New` pop-up. Secondly an alert function - which checks for a release-specific HTML/MD file in a directory of the repo and if a file is there it displays it in a pop-up - this would allow you to alert users to critical bugs in specific releases. Thirdly a Current Release HTML/MD file which contains a **user-friendly** description of the functionality in a new release, and which is downloaded and shown as a pop-up when the release checking shows that there is a new release available.

tatarize, 2023-04-29:
It should be fixed in the latest beta. There's just some herding of cats on a full release.

Sophist-UK, 2023-04-29:
As I have said multiple times - but apparently no one is listening - the evidence suggests that the symptoms are extremely unlikely to be caused by USB errors because in the original pictures above it is **not** losing absolute position., but instead is burning part of some subpaths in the wrong place and then regaining position, and because there appears to be a scaling issue, neither of which are likely to be due to packet loss.

Also, since it 0.8.2 burns reliably and 0.8.3 unreliably, it is highly unlikely to be an issue with code that was in earlier versions of 0.8 which I have been using in production up to and including 0.8.2 with great reliability. (TBH considering what has been discovered with the USB stuff, I am surprised that these versions have been as reliable as they have been.)

There is clearly another bug - and since these symptoms were not evident in Jens extensive and time consuming testing, I suspect that they are something to do with my settings being different - which is why I posted pics of them above.

However, next time I am in the workshop I will try the new beta and see if it immediately has the same problem, and if not I will keep using it to see if the problem is intermittent.

Is there anything you want me to do to improve diagnostics if I try it again with the new beta and it fails again?

Sophist-UK, 2023-04-29:
P.S. Given that the 0.8.3 full release seems to have been somewhat premature given #1782 (which with hindsight did not provide you with enough information / evidence and where the raster fix was clearly not the solution to the issues I was having but I didn't call it out), I would suggest that you wait until we have some reasonable proof of stability with betas before a full release, and that if you want to release new functionality it should be via another beta until you are sure that the code is stable.

tatarize, 2023-04-29:
It's actually a transmission issue. The USB error packets are not recovered correctly. A 207 error would lose the packet completely rather than have it resent. Which looks like it's losing the positon but it's just losing a packet of movement info.

Sophist-UK, 2023-04-29:
Yes - but this lost packet of movement information results in an absolute and unrecoverable loss of position, and since USB errors are random, it is highly unlikely that these would cancel each other out. And even if they did, the distance that the partial sub-paths are misburned by and the fact that the position was recovered afterwards (albeit with scaling issues) suggests that this is **not** due to losing "packet movement info". But as I say I will try the new beta and see if it recurs.

tatarize, 2023-04-29:
Tests were done correctly since it was really really hard to track otherwise. I ended up feeding my laser 30% bad checksum packets to see how the recovery went. Jpirnay had a usb hub that managed to throw enough errors that we knew it was real and had to come up with some methodologies to test things. The fix, did fix things. It's totally a USB transmission error recovery thing.

tatarize, 2023-04-29:
Not quite, all the movements of the laser are non-absolute. It's entirely relative so it would lose some movements. They could cancel each other out, but random flying in 2 dimensions only has a 33% chance of returning home. And it's not really a lost packet but likely some kind of error processing it or checksum issue in the card. I would expect it got there intact, but sends back a 207. This means it needs another copy. And failing to send the copy basically drops packets out of the work. And yeah, that smears thing and does vectors all screwy.

Sophist-UK, 2023-04-29:
Apologies if my comments were in any way suggesting that you hadn't found a serious bug - clearly you did - and both you and Jens put in a **LOT** of effort to fix this. But the evidence from this testing is more like the first issue in #1782 (which was a shift) - but no evidence in Jens testing of the issues reported here having been reproduced.

tatarize, 2023-04-30:
Bug fixed.
0.8.4 released.
0.8.3 nuked

Sophist-UK, 2023-04-30:
So - all my logical points about how this may not fix the problems reported in this issue are not worth even responding to?

And without having had feedback from me about whether it has fixed them, you have (yet again) gone for a full release rather than a beta with the risk of further reputational damage if this "critical" issue (YOUR WORDS) is still not fixed.

Sophist-UK, 2023-04-30:
Well, some good news. 😄 

I tried 0.8.4 then 0.8.3 then 0.8.4 and it appears that the issues I was experiencing are indeed fixed  by 0.8.4.

I am not sure whether this is supposed to happen, but I am still getting Rejected Packets. I load MK 0.8.4, I click home and this is the resulting controller window...
[image]
As you can see the packet count is ridiculously high, and apparently there were **110 rejected packets* (though whether this is true or not would seem to depend on whether it also starts from a non-zero value).

When I burn the file, I get no further rejected packets. When I do additional homes, I don't get any further rejected packets either. I was not able to reproduce the Home grinding issue in the limited testing I did.

When it starts to raster I get `[Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'` in the USB log.

So this looks at least as stable as 0.8.2 and I will be using it in production.

closed 2023-04-30 by Sophist-UK

tatarize, 2023-04-30:
The rejected packets are fine. The problem is that the 0.8.3 version was not dealing with the rejected packets correctly. Now it does. You had a bunch of rejected packets before it just couldn't deal with them correctly, and would break. Some folks don't have any and it wouldn't have broken making detection a bit harder. But, we ran down the stuff it could be since nothing changed the code being sent, it had to be transmission. Only jpirnay also had some rejected packets and ran down what it could be and came up with additional testing routines.

The timeout is a timeout error. You got that because you had a packet timeout. I'll take it under advisement, it's pretty common and isn't a problem. Sometimes the card takes to long to reply. But, it's not an issue.

Sophist-UK, 2023-05-01:
As I said, I don't get **any** rejected packets during a burn now. But I get 100 on initialising the controller. And the packet count does not appear to start at zero.

The packet timeout occurs consistently on starting a raster and is IMO not random.

Sophist-UK, 2023-05-02:
And I still do not understand why this worked in 0.8.2 and stopped working in 0.8.3b3. And loose ends like this niggle.

Sophist-UK, 2023-05-03:
I am not sure what would make a packet rejected, but it definitely seems odd consistently to get rejected packets during initialisation, and you have to wonder whether this is permanently sending incorrect packets.

jpirnay, 2023-05-03:
This is a lifetime count, not a session count, these values get persisted. I assume I add a reset option to it.

Sophist-UK, 2023-05-03:
I am not sure why these counts are persisted. Does this actually make sense?

tatarize, 2023-05-03:
Yes, because it tell you how many times you've gotten these things. I guess you could lose that information or have it flagged with a _ but it's generally kind of useful to know. You haven't had bad packets in x amount of time.

Most people won't know what they are, but if we want that information we don't want to know if they channel has been noisy lately, not this session.

Sophist-UK, 2023-05-03:
Then don't we want to save information by session, and maybe save for the last x (10? 20?) sessions? And somehow tie it up with what was burned.

If we believe that rejected packets should now be a rare event,. should we instead save details (in a log file) of rejected packets and something about what those packets were.

But I guess, if they are rare and we have a console command to reset them, that is probably good enough for me.

Sophist-UK, 2023-05-04:
These packet values are NOT visible in settings and so cannot be reset using a `set` command.

Jens was talking about a console command to reset them (which is probably about 5 lines of code).

tatarize, 2023-05-04:
They are, but only in the device, not in root. Like `set -p lihuiyu ...` or something like that. Try `context` to see the other relevant contexts.

Sophist-UK, 2023-05-04:
That makes sense.