meerk40t/meerk40t #1813: [Enhancement] Multi-pass order with differing op passes

Sophist-UK, 2023-04-20:
Suppose I have an inside cut with two passes (small holes in acrylic) and an outside cut with  one pass - and I don't use Merge Passes because it would cut each small hole twice in quick succession and I want to leave time for the acrylic to cool between passes. But I do have Burn Inner First

MK does inner cut, outer cut, inner cut.

I would like to suggest that instead of doing a reducing number of operations in each pass, it starts by doing only the ops with the greatest number of passes, and then adding more and more ops until the last pass includes all ops.

tatarize, 2023-04-20:
You don't have merge passes. So the passes are confined to their own group. Which then only is allowed to merge operations within that space. There is no checkbox for being allowed to move the entire operation, especially since you could have a beep or something else in there, and without merge passes these are a fairly hard barrier.

I get the point, I don't see an easy solution.

Sophist-UK, 2023-04-20:
The solution is easy. Rather than iterate the ops decreasing the pass count after each pass and burning those whose pass count is still >0, instead you iterate the ops and collect the maximum passes, and then you iterate backwards burning those who passes >= the iteration count.

tatarize, 2023-04-20:
With merge_ops and not merge_passes set, we call the `_to_blob_plan_passes_first()` function. I divided these functions up so they are more each stage gets a single self-contained function, well named and well documented:

```python
    def _to_blob_plan_passes_first(self, grouped_plan):
        """
        If Merge operations and not merge passes we need to iterate passes first and operations second.

        This function is specific to that case, when passes first operations second.

        Converts the operations to cutcode.
        @param grouped_plan:
        @return:
        """
        for plan in grouped_plan:
            pass_idx = 0
            while True:
                more_passes_possible = False
                for op in plan:
                    if (
                        not hasattr(op, "type")
                        or op.type == "util console"
                        or (
                            not op.type.startswith("op")
                            and not op.type.startswith("util")
                        )
                    ):
                        # This is an irregular object and can't become cutcode.
                        if pass_idx == 0:
                            # irregular objects have an implicit single pass.
                            yield op
                        continue
                    if pass_idx > op.implicit_passes - 1:
                        continue
                    more_passes_possible = True
                    yield from self._blob_convert(op, 1, 1, force_idx=pass_idx)
                if not more_passes_possible:
                    # No operation needs additional passes.
                    break
                pass_idx += 1
```

This yields from self._blob_convert()

```python
    def _blob_convert(self, op, copies, passes, force_idx=None):
        """
        Converts the given op into cutcode. Provides `copies` copies of that cutcode, sets
        the passes to passes for each cutcode object.

        @param op:
        @param copies:
        @param passes:
        @param force_idx:
        @return:
        """
        ...
```

So currently we iterate through our passes, 0 to some number of passes. If we have pass 0 we go ahead and yield that op. We also yield, yield our ops for everything that still has remaining passes.

So in our regular example we do: Pass0-Inner-Cut, Pass0-Outer-Cut, Pass1-Inner-Cut. We do not know what the max cut actually is since we're just iterating our passes and setting the forced_idx on the _blob_convert() higher and higher. In theory we should not ever group an irregular and normal object together. Our regular groups should be merged.

If we, instead queried to find the highest op.implicit_passes and then iterated backwards:
* Pass1-Inner-Cut
* Pass0-Inner-Cut
* Pass0-Outer-Cut

---

Logically this hits some would-be problematic workflows.

Say I want to use this because I might not need all the passes for some part. I want to set my cuts such that I think 10 passes will cut it at the end. So I set these settings. My main part of the work engraves 1 time. Then it it starts cutting the passes to actually cut the material. --- This is a decided useful workflow.

However, if we implemented this, it would start by cutting 9x times through the material and then only on the last cut would do the entire worksheet.

* Pass(N)-Cut
* Pass(N-1)-Cut
* Pass(N-2)-Cut
* Pass(N-3)-Cut
* Pass(N-4)-Cut --- Part is cut falls out.
* ....
* Pass(1)-Cut
* Pass(0)-Main Work
* Pass(0)-Cut

The current ascending passes would have done that

* Pass(0)-Main Work
* Pass(0)-Cut
* Pass(1)-Cut
* Pass(N)-Cut
* ....
* Pass(N-1)-Cut -- Part falls out.
* Pass(N-2)-Cut (ABORTED)
* Pass(N-3)-Cut
* Pass(N-4)-Cut

Sophist-UK, 2023-04-21:
Yes - that is definitely an issue to be avoided. I guess with Cut inner first, you need to do a hybrid where you count down excess non-cuts until the last non-cut pass where you include the first cut pass and then count up for the cut passes.

tatarize, 2023-04-21:
The code effectively breaks each pass into a layer. You run each layer as it's own group of elements that must be done during that pass. So each blob_group is a layer.

You have as many of these layers as you have the maximum number of passes.

Each operation needs to run `p` times and our current code says each layer runs `0-p`, I could see maybe needing some of these to switch up a bit and wanting more control over them. And each element either runs in the pass or doesn't run in the pass. But, I don't see any easy way to decide this. And stacking the lower pass items first seems superior to putting them last.

In theory, I'll be switching away from cutcode and towards geomstr cuts and it might be reasonable to divide those into layers in the job. It's basically going to be the same sort of thing.

If we have merge operations and merge passes, we get: 1 layer.
If we merge operations, but do not merge passes, we get: `max_passes` layers.
If we do not merge operations but merge passes, we get: `operation` number of layers.
If we do not merge operations and do not merge passes, we get: `operations * passes` layers.

There are a few other places where we effectively get layers, like zdepth objects and others.

---

Imagine the original routine with 2X cuts for the inner and 1X cuts for the outer, `Merge Passes=False, Merge Opertions=True, Cut_Inner_First=True`.

The current code still does `inner, outer, inner`. This may actually be contrary to the setting for `cut_inner_first`. If I know that the `outer` is outside the `inner` and that inner must be cut first. Regardless whether I'm merging passes I should maybe complete all the passes of cut_inner before I complete the passes of cut_outer.

If we slice this into layers. We get:
Layer 0: Inner, Outer
Layer 1: Inner

But, we were told to cut the inner most before the outer-most so we might need to do that as:
Layer 0: Inner
Layer 1: Inner, Outer

We were instructed to merge operations but not merge passes. This seems like it reverts the problem to a question of constraints. Though I suppose here the constraint that inner happens first actually divides that layer too and prevents the merging. Within a layer we can reorganize any part cuts to minimize moves. And we can't do that for `Inner, Outer` if we're specifically insisting that inner occurs first. This gives us layers instead:

Layer 0: Inner
Layer 1: Inner
Layer 2: Outer (Must occur after all layers containing inner).

I believe that currently and effectively `cut-inner-first` means that any layer which contains nested layers is divided into inner-layer and out-layer, though this is more complex since you could have a series of inner and outer layer with A and B in C where you can finish A or B and then complete C, and optimize as much as you want within that hierarchy. However, secondary passes of inner, would still currently occur *after* outer. And it also gets a bit more complex when considering something like 2x passes of outer and 3x passes of inner. Your required order, in that case would be `inner inner inner, outer outer` rather than the current setup which would give you `inner outer, inner outer, inner`.

I don't know if this is an extreme edge case or if there's a better solution to do during a partial rewrite of that code when it gets converted (a bit) over to geomstr.

closed 2024-10-04 by jpirnay