meerk40t/meerk40t #1829: [0.8.2] Simulate & Burn only burn a small part of project

Sophist-UK, 2023-04-23:
[image]
Simulate take a long time to optimise and then shows this:
[image]

Simulate seems to be considering all the elements given the time it takes, but then only simulates/burns the first path. Actual burn is similar. Unchecking Group Inner burns makes it work.

Here is the file:
[tinydoctorsvg x4 195x205](https://user-images.githubusercontent.com/3001893/233854850-5de68a2c-33be-43f9-9430-dafd39feb016.svg)

jpirnay, 2023-04-24:
Confirmed in my setup in latest main:
No 'burn inner first':
[image]

With 'burn inner first' (regardless of group inner)
[image]

Sophist-UK, 2023-04-24:
Not sure I see the simulation differences here. But glad you could reproduce it.

tatarize, 2023-04-27:
The `path_to_cutobject` code ends up creating empty cutgroups because they end like:
`.. 6938.70909366,7085.3123003 6949.61300304,7085.3123003 C 6960.5029688,7085.3123003 6969.35716631,7076.4579264 6969.35716631,7065.5537998 Z M 6986.41021129,7065.56774369`

This does Z (close path) then a singleton M (move). And when checking candidate cutgroups these empty cutgroups remain candidates. Move isn't cut and these are not empty (they contain a move), and since there are candidate groups, largely because they aren't burned yet because there's nothing in them that can be burned. They remain candidate groups that can't burn.

If there are no candidates groups, then we're supposed to pick from all viable candidate groups, but we do have groups candidates in the form of unburnable empty groups. We are not burning a path segment here, and this isn't a closed path, so we yield all the segments inside these empty groups (read no segments).

Since we gave all the candidates within the permitted candidate groups and no candidate was valid, we are done. All candidates were determined to be burned. But, this logic is entirely nonsense since empty candidate groups, don't contain segments and therefore we're in a situation where we have no potential candidates left and we should have fallen back to opening up a new layer of burning.

Sophist-UK, 2023-04-27:
This wasn't a specially crafted SVG, but rather one created and saved in Inkscape. 

I suspect that a few years ago it started life as a vector PDF and I imported this in Inkscape and played around with colours, line widths, groups, compound paths etc.

I guess the lesson to be learned here is not to trust external input (e.g. SVG files) and to discard stuff that is semantically empty (like this) - I am unsure whether this should be discarded on load or on burn (though it probably makes no difference to this issue).

However, from a perspective of code quality, MK should not discard internally created wrong data but should instead throw an exception so that the bug can be tracked down and fixed - this plus good unit tests are the route to reliable code. And on this basis, perhaps the empty path should be discarded on load.

tatarize, 2023-04-28:
Yeah. was a bug. SVGs aren't trusted. It just narrowly managed to cause a series of narrowly defined edge conditions.

closed 2023-05-02 by tatarize