meerk40t/meerk40t #1831: [0.8.x] Abort mostly grinds the steppers by hitting end stops

Sophist-UK, 2023-04-24:
When I use the Abort button in 0.8 the burn aborts but rather than just stopping immediately and killing the instructions queued inside the m2, it instead moves and often grinds the steppers.

Sophist-UK, 2024-10-04:
Not completed. Still grinds in 0.9.5beta.

jpirnay, 2024-10-05:
Unclear whether the queue / m2nano status can be changed as requested, code is rather clear about it:
```
    def reset(self):
        """
        This command asks that this device be emergency stopped and reset. Usually that queue data from the spooler be
        deleted.

        Asks that the device resets, and clears all current work.

        @return:
        """
        self.service.spooler.clear_queue()
        self.plot_planner.clear()
        self.spooled_item = None
        self.temp_holds.clear()

        self.service.signal("pipe;buffer", 0)
        self(b"~I*\n~")
        self._reset_modes()
        self.state = DRIVER_STATE_RAPID
        self.paused = False
        self.service.signal("pause")
```
I don't see any delays in clearing the queue.
Will go the backlog