meerk40t/meerk40t #1859: [Bug]: Gcode Board not working

dorsi90, 2023-04-30:
### Summary Description

Hi dear, with latest release gcode board not working anymore.
The only command that send and return a response in controller panel is the clear alarm button, other commands or actions not working.
I switched back to meerk40t 0.8.2000 and all actions works but with this version i can't see the serial data sended and received in controller panel which i see in 0.8.3 but i can't find this version anymore on github.

### MeerK40t Version

0.8.4000

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-04-30:
Drat. It got modified a bunch but was expected to work. I'll run it up the flag pole. There's some things like pausing and resuming that the device sucked with that.

In theory if you run from source you could check out 0.8.3000 tag. Though it had a critical flaw in the lihuiyu driver. So it got nuked since I didn't want anybody using a broken version.

dorsi90, 2023-04-30:
@tatarize yes, pause and resume not working all the times but there are other bugs that cause the application to crash for example if i turn off the machine meerk40t not close the serial connection then when turn on and try to disconnect and reconnect the application crash.

Tomorrow i try the 0.8.3 from source that is a little bit better then 0.8.2 because i can see the serial data transmission.
Thanks a lot

tatarize, 2023-04-30:
What exactly is wrong for 0.8.4. We can't seem to figure it out. What version of Grbl are you running and how does it fail. It should be pretty minor but we can't seem to make our devices fail.

In theory there should be a couple channels `send-{comport}:{baud-rate}` and `recv-{comport}:{baud-rate}` that should be accessed in console with the `channel` command. But, I'm not sure where the issue could be since the testers seem like they work fine.

The serial transmissions is a bit sketchy. Also the TCP transmissions is too different for the TCP controller. So that stuff does need fixing.

tatarize, 2023-05-01:
Okay. We went through all of GRBL and found all the bugs we could find that might have broken anything including all the parts I mentioned above.

This should, quite hopefully be significantly better for most uses.

https://github.com/meerk40t/meerk40t/releases/tag/0.8.4001

dorsi90, 2023-05-01:
@tatarize i tested the executable and from source version 0.8.4001 and not working.
When i open the application it's already connected but all actions not working. Starting from version 0.8.4 there is "controller buffer" maybe this cause the issue.
[image]

If i try to connect and disconnect nothing happen and only clear alarm appears to send and receive the serial data.
[image]

I've tested the version 0.8.3 from source and works very well
[image]

dorsi90, 2023-05-01:
@tatarize i upgrade the firmware of my grbl controller from fluidnc 3.6.8 to 3.7.0 and now works with meerk40t 0.8.4001.
So maybe this issue is related to older grbl versions or just with the specific version 3.6.8 however fluidnc 3.6.8 is only two months old

dorsi90, 2023-05-01:
After more tests it keeps malfunctioning, sometimes it connects but the laser doesn't move across all the plane other times it doesn't connect at all.
I also continued to try version 0.8.3 and it continues to work very well so i don't think the problem is the grbl version :(

tiger12506, 2023-05-01:
Our lasers that we test with are GRBL 1.1f (x2) and Marlin (something). I don't think any of us has fluidnc to test against. You should join the MeerK40t Discord server and we can probably make the modifications necessary to get yours working properly.

dorsi90, 2023-05-01:
@tiger12506, yes it works in 0.8.3 and below.
FluidNC is 100% compatible with gcode of GRBL 1.1f, it's an upgrade of Grbl_Esp32 the noticeable difference is command $$ that return only $10 so we can't read the settings with this command that can be useful for estimate the time in simulation.

tatarize, 2023-05-02:
The 0.8.5 beta-1 is mostly GRBL corrections over 0.8.4. We've not really gone through the various versions to know exactly which features are permitted and not. But, testing them on several different devices this current iteration seems to generally work.

tatarize, 2023-05-02:
If you connected to the device raw would it give you any greeting message. Like during bootup? If so what? I see from your connecting that it says "connecting to grbl" but never confirms the connection like it never gave the typical this is grbl 1.1f message at the start.

tatarize, 2023-05-02:
```
[MSG:INFO: FluidNC v3.6.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.21 MOSI:gpio.22 MISO:gpio.23]
[MSG:INFO: SD Card cs_pin:gpio.19 detect:NO_PIN freq:0]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (10.000,3510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.32 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.17]
[MSG:INFO:  X Pos Limit gpio.5]
[MSG:INFO: Axis Y (10.000,3510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.4]
[MSG:INFO:  Y Pos Limit gpio.16]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.14:low Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.15]
[MSG:INFO:  Y2 Pos Limit gpio.2]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.12 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:TP-LINK_8452]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.107]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

IF that's the startup routine and id doesn't actually says it's using grbl it may not allow the connection.

There was a bug where it would, if not told the machine was booted, send data to the controller before it finished booting up which would cause issues. Now it waits for `"Grbl *"` to say that the device is booted. This appears to be a related firmware but not actually GRBL and we may need to make adjustments.

tatarize, 2023-05-02:
https://discord.gg/n6Yp6ZEw

For the discord channel. We're not quite sure what the difference is, but we're thinking it's something minor. It really should confirm the connection but maybe it would. The GRBL interface thing made it clear that was the indication it was booted. 

> `Grbl X.Xx ['$' for help] :` Welcome message; indicates initialization.

So I assume it sends that message it wouldn't really let you send any data if it didn't.

The controller buffer there is typically for character counting. You could switch to sync from buffered and it probably would stop mattering at all. Since really that's just a question of how much extra data ahead should the data creation be from the data sent. Basically you could send all data to the controller right away but you'd not really know the position or a lot of the other information since you'd just feed the entire gcode file to the memory of the controller to send that when it needs to.

In theory grbl says you can send a `G4 P0.01` and it will force a resync of the planning buffer and CPU buffer. This should mean that it's not storing a bunch of other things in the planning buffer. But, I'm not entirely sure the nature of the incompatibility here or what would be needed to overcome it. 0.8.4001 is much better grbl code, though it doesn't fully work and it's not clear why.

In theory the controller should contain all the data sent and recieved and probably give some reasons why it's not working. In theory I guess we could add some events for some common potential issues. For example, Buffer suspended because connection is not confirmed (Never said "Grbl..."), but it's not quite sure what the issue here is if it's also not quite sure why it's not working.

If you blame the Control Buffer, turn it off. De-check `Limit the controller buffer size` and it'll be unlimited which was the pre-8.3 behavior.

dorsi90, 2023-05-02:
@tatarize on boot there is a welcome message but not identical to "Grbl X.Xx ['$' for help] :". This is not a problem because i can change the welcome message on boot like that and later i try it.
The problem is that if i connected the machine already booted there isn't any more welcome messages on every connections.
Later i try to force to display the welcome message on every boot but maybe for a better compatibility you can send a command on connection to get borad info and check if the string returned contain the keyword "Grbl"

I think the problem is this line of code in grbl/controller.py added on version 0.8.4
elif response.startswith("Grbl"):
                self.grbl_events("Connection Confirmed.")
                self._connection_validated = True

Later i make some tests and give you more feeds
Thank's for your work

tatarize, 2023-05-02:
I can make a version that can and should make that intro text configurable. It's simply possible for meerk40t to try to send data before connecting and connect during the process but, that present data that it couldn't actually run so that bit of code in there was added to prevent that bug.

You could go and find the `self._connection_validate` variable in the init and set it to `True` which would just ignore that. Or change the `response.startswith("Grbl"):` to whatever the actual welcome message is, so it would still look for that but find whatever version your laser uses.

closed 2023-05-07 by dorsi90