meerk40t/meerk40t #1870: [0.8.4] Unexected laser start and crash

Sophist-UK, 2023-05-02:
Selected Elements node, went to position pane, clicked and entered X value, tabbed and entered Y value, pressed enter.

Laser started a burn and MK had a crash:
```
MeerK40t crash log. Version: 0.8.4000 on Windows: Python 3.7.9: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "meerk40t\kernel\kernel.py", line 1793, in schedule_run
  File "meerk40t\kernel\jobs.py", line 95, in __call__
  File "meerk40t\kernel\context.py", line 322, in console
  File "meerk40t\kernel\kernel.py", line 2150, in console
  File "meerk40t\kernel\kernel.py", line 2205, in _console_parse
  File "meerk40t\kernel\functions.py", line 279, in inner
  File "meerk40t\core\planner.py", line 670, in plan_clear
  File "meerk40t\gui\busy.py", line 92, in change
  File "meerk40t\gui\busy.py", line 119, in show
AttributeError: 'NoneType' object has no attribute 'SetClientSize'
```

jpirnay, 2023-05-02:
I need to separate this for my understanding:
- meerk40t started a burn without you asking for it
- then the crash happened
(that crash should not happen but has been contained now). More concerning is the first part. Have you the laser arm option active or is it disabled?

[image]

Sophist-UK, 2023-05-02:
No - I don't have Laser Arm enabled.

I suspect that despite having clicked inside the X box of the position pane and tabbed to the Y box, the Enter key was someone linked to the Green immediate-burn button either the stand-alone pane or the similar button in the Laser panel.

tatarize, 2023-05-02:
Yeah, it's a bug. `self.frame` starts at None.

The frame is set during `self.start()` but `self.show()` assumes it exists and tries to set the client size without confirming the window exists.

tatarize, 2023-05-02:
Jpirnay fixed most of it, I think he missed `self.text` tho

tatarize, 2023-05-02:
I picked up the spares there. That should not crash.

tatarize, 2023-05-03:
I'm pretty sure you could tab and enter to something that starts the job on accident. It's why none of those exist by default. You need to disable laser-arm to have one for the panel and you need to add the "go" button.

Sophist-UK, 2023-05-03:
Personally I don't think a burn should start unless the user has clearly decided they want this to happen i.e. only under the following circumstances:

1. A click on an execute button or menu item. (I think that preventing this single action to burn is the main purpose of the Laser Guard.)
2. Keyboard selection of an execute menu item (which takes a sequence of 2 or more key clicks).

If you ever want the Enter key to immediately start a burn, then I think it should only be when the user has tabbed (if indeed this is possible) to give focus to the BGBB (big green burn button), and certainly **not** when the focus is in e.g. the Position Pane and it will be unexpected.

I see these rules as being safety-related and do not see these rules as being over burdensome to users.

jpirnay, 2023-05-03:
Addressed in [PR 1873](https://github.com/meerk40t/meerk40t/pull/1873) - will only accept a conscious mouse click

tatarize, 2023-08-26:
The busy crash turned out to be a bit more involved, and just setting the in try/excepts wasn't really correct; it needed to have a sync lock which it now does for all branches.

closed 2023-08-26 by tatarize