meerk40t/meerk40t #1875: [0.8.4] Burn inside first does not work on attached design

Sophist-UK, 2023-05-04:
[ruler](https://user-images.githubusercontent.com/3001893/236233285-ef93b9f5-24ba-462f-a9e8-e488cd0cce30.svg)

Sophist-UK, 2024-10-04:
If you are going to close bugs which are fully documented and reproduceable, then that is just crass.

jpirnay, 2024-10-08:
If you are reporting a bug that is your own design flaw, then this is ... ?

The svg file you attached shows that the outer edge is actually constituted by two paths:
```
<path d="M 25.5,50.0001 L 178.466,50.0001 L 354,50.0001 L 354,1.00008 L 178.466,1.00008 L 25.5,1.00003 M 25.5,1.00003 C 11.9272,0.999982 0.999574,11.9271 0.999574,25.5001 C 0.999574,39.0733 11.9272,50.0001 25.5,50.0001" fill="none" fill-rule="evenodd" id="path368" inkscape:label="Edge" label="Edge" lock="False" stroke="#ff0000" stroke-linecap="round" stroke-linejoin="miter" stroke-width="0.2" stroke_scale="True" transform="matrix(2197.5572744387373, 1351.9462555525183, -1351.9462555525183, 2197.5572744387373, 48801.940976718906, -24665.162944409552)" type="elem path" />
```
relevant is the ``L 25.5,1.00003 M 25.5,1.00003 C 11.9272,0.999982 0.999574,11.9271`` the **M 25.5, 1.00003** - this the SVG indicator that a new subpath starts. By definition neither of the two subpaths are closed. So the check inner algorithm will not consider these two subpaths of the outer edge something that needs to care about inner objects. If you look to the simulation then you will see that the stitch point between the line segment and the circular segment will be used to continue the burn.
[image]

If you remove this "M" command from the path definition (corrected file attached below), then the path is a contiguous path and things are working as expected.

[image]

[image]

closed 2024-10-08 by jpirnay

Sophist-UK, 2024-10-08:
Ooops - well that is embarrassing - if the Edge "cut" is actually two unclosed sub-paths, "of course" it is not going to work.

That said, with this as an example, perhaps MK should attempt to join two subpaths with the same end-points in an attempt to create a single closed path (in the same way that it closes an open path that has coincident ends but which is technically not closed). I accept that this idea is A) debatable as to whether it is sensible; B) how to do it when there are more than 2 end-points co-incident; C) what potential incorrect consequences might ensue.