meerk40t/meerk40t #1886: [0.8.4] SVG load of Hidden Text objects causes pop-up

Sophist-UK, 2023-05-11:
I have an SVG file which has text objects which are hidden - because I duplicated them and converted the duplicates to paths (keeping the original text objects in case I want to edit them and reconvert to paths).

However despite all Text objects being hidden, when I load the file I get:
[image]

In Preferences/General I have:
[image]

IMO this warning should only be shown if there are visible features and hidden features should be ignored.

tatarize, 2023-05-14:
Hm. That does seems to be a bit less than useful. I see why it would do that though. I think if rather than hidden you set them to disabled it might not load them at all. In theory it removes it completely from the render tree which has the result of not parsing them. Svgelements by default will access hidden but won't even touch disabled or whatever. Though I think a flag to let you do that got added a year or so back.

Doesn't really fix this issue. I guess we'd just check for the post-process and not perform it on hidden objects. I don't see a clever way other than checking.

Sophist-UK, 2023-05-19:
I am getting this pop-up with "clipping" as the reason when loading a design which has no clip paths.

Sophist-UK, 2023-06-10:
I just got a similar warning about "clipping" and "mask" for a file which doesn't use clipping or masking.

I suspect that there are some clip or mask defs (in the defs xmls section) even though these are not actually used.

tatarize, 2023-06-11:
The difference between hidden and disabled is generally that the render tree isn't invoked for the latter, which actually means it won't show up for svgelements.

I assume this is loading and running through inkscape. I'll have to look at that some more. I'm not really sure when that kicks on and the rest of the code needs to be checked. It worked and was off by default so I didn't check much beyond that since it was somewhat defacto harmless.

I was pretty sure it checked for unsupported elements or something during loading but I'll add it to the list of things to check after I get 0.9.x more fully working

Sophist-UK, 2023-06-11:
I am not sure how these checks for unsupported SVG elements are done, but I suspect that they are done when such things are read rather than when such things are used.

Because I know what MK can and cannot handle, I generally click the `Load Original` button and the file loads fine, so I do NOT load it through Inkscape using the `Convert` button. My point is not that there is a bug, but rather that this warning pop-up has false positives.

tatarize, 2023-06-11:
Okay, the `inkscape.py` routine in extras, registers a `check_for_features` routine as a `preprocessor./.,svg` in the kernel. This is always checked during the `elements.py/load()` routine. And merely executes, taking the pathname and returning *potentially* a new pathname. This API seems pretty solid. 

So the important routine is just the check_for_features, https://github.com/meerk40t/meerk40t/blob/acd0a39a03cd97b9790e5c7667823d17072fe46f/meerk40t/extra/inkscape.py#L384

The routine to check for these features isn't an XML parser. It's actually just searching for the words "`<text`", "`<clipath`", "`<mask`" or "`<lineargradient`"/"`<radialgradient`" anywhere in the svg.

```python
            features = {
                "text": [False, ("<text",), METHOD_CONVERT_TO_OBJECT],
                "clipping": [False, ("<clippath",), METHOD_CONVERT_TO_PNG],
                "mask": [False, ("<mask",), METHOD_CONVERT_TO_PNG],
                "gradient": [
                    False,
                    ("<lineargradient", "<radialgradient"),
                    METHOD_CONVERT_TO_PNG,
                ],
                "pattern": [False, ("<pattern",), METHOD_CONVERT_TO_PNG],
            }

            with open(source, mode="r", encoding="utf8", errors="surrogateescape") as f:
                while True:
                    line = f.readline().lower()
                    if not line:
                        break
                    for feat, entry in features.items():
                        for candidate in entry[1]:
                            if candidate in line:
                                entry[0] = True
                                if entry[2] > needs_conversion:
                                    needs_conversion = entry[2]
```

If it finds that it sets the given method, and performs the given call.

So your file apparently contains the word "<text" or "<clippath" or "<mask" somewhere in there. That seems about right. And we should, rather than parse the file line by line, parse it tag by tag, and turn on if we detect these otherwise disallowed functions and calls them.

So the reason the hidden stuff will still cause the pop-up is because it searched the file and found the relevant tags. This would actually *still* trigger regardless how it's processed. Even if you set it to disabled this would process since there is no valid tag here.

tatarize, 2023-06-11:
To avoid this issue, we'd basically need to parse the svg before we parse the svg. This seems prohibitively hard for a supplemental feature.

Sophist-UK, 2023-06-11:
It seems to me that there might also be a logic issue in this and some usability issues:

1. LOGIC: If the file contains both `"text"` and one of `("clipping", "mask", "gradient", "pattern")` then you need to do CONVERT_TO_PNG (which is bruter than CONVERT_TO_OBJECT), but whatever comes last in the file will be the single conversion done.
**Note 1:** The Inkscape action is to convert the TEXT OBJECT to a path - so the option should probably be named CONVERT_TEXT_TO_PATH.
**Note 2:** The pythonesque approach would be to have separate methods / functions for `convert_text_to_path` and `convert_to_png` and to assign the method / function object to a variable and then later call that method.

2. USABILITY: Convert (only) text objects to paths is relatively benign to the burn because they will remain as paths with fill which will still be classified in the same raster operation(s) whereas converting the entire design to a PNG potentially screws the entire burn because the whole thing will be rastered including everything that would previously have been vector burned (and it will be in an Image operation rather than the existing Raster operation). IMO the difference between these is so great that they need to be presented to the user differently - and indeed have more preference options to e,g, automatically convert text objects to paths but not automatically convert the entire design to PNG.

I am assuming that there is not an Inkscape option to convert only elements with clippaths, masks, gradients or patterns to PNGs? whilst leaving everything else as vectors and this is why you have to convert the entire design? Also, even if you can convert only these elements to images, the clippaths etc. can be applied to things that are vectors rather than images, and so the above comment about it being a change with massive implications would still hold true.

tiger12506, 2023-06-11:
That *is* what parsing is. A simple search is one thing, but when you're tracking hidden/disabled and tracking parentage of hidden/disabled -- that's literally parsing.

Regardless, Tat marked it accepted. I agree--those are good points you make.
I would assume that if we did parse the svg, then we could selectively send certain elements to inkscape to be converted.
So the question is -- can we delay this check until the svg is parsed?

tatarize, 2023-06-11:
Yeah, literally to figure out whether a branch would have a visible element it would require parsing the data. Currently the parse looks for tags `<clippath` et al, by a simple line-by-line check. To see where it falls in the tree with the given property would require that we do basically a normal svg parse.  We could do that with svgelements.

It's marked as accepted since it's at least clear what's going on, and the original issue is markedly true.

The code is set in a pre-processor, so really it would need double parsing or the change the api to keep the objects for the initial parse, or call inkscape and replace them with a new parse after that.

But this seems harder, it seems like we'd be better off just parsing it twice anyway.

This is a edge condition issue. I guess we could also just add in a custom xml parser and parse it again, though it wouldn't detect things like if you have a Pattern in a `<defs>` but then never use that def in a `<use>` block it would fail. The only real surefire way would be to parse it, then is if we find a tag that we're not allowed (though `<clipPath` is actually somewhat supported, it's not just supported in meerk40t). 

To avoid parsing it twice, we'd need to change the API to not do a preprocessor, but rather a sort of post-processor that is smarter as to the type of file we're parsing. It needs to parse the file, get all the tags. Then we check through all the tags we have looking for these particular non-supported one, and then abort the parse in progress (or after we've checked everything and know we need some changing) and parse it again. So most would get 1 parse and a review. And some would get 1 parse, a review, and another parse.

This seems like its rewriting the feature to fix a rather edge condition defect, that it detected the tag but didn't check whether that tag was properly used.  And the current form of the feature is really nice. It's always done for any preprocessor script, and completely agnostic as to what type of file is being read.

closed 2024-10-04 by jpirnay