meerk40t/meerk40t #1895: [0.8.4] Suggestion: Move head to click point improvement

Sophist-UK, 2023-05-14:
In order to increase head placement accuracy when using the click-to-move functionality (after right-clicking in the move box), can I suggest the following...

If there is a path node in any of the selected / emphasised paths (or all paths if none are selected) within a certain screen distance of the click point, then move to the closest path node point - if no such nodes, then move to the click point itself. Screen distance is suggested (rather than bed distance) because that is related to the accuracy of mouse usage.

tatarize, 2023-05-14:
So snap-to-element for position tool?

If there's no nodes, where would it move the click-point to?

I think it actually does work. It didn't show the field but it clearly snapped to grid, and would snap to elements when I held shift. (I have snap-to-elements off so shift inverts it), and clearly snapped to node points.

Sophist-UK, 2023-05-15:
I have checked my Scene Preferences.
[image]

P.S. I think the tooltips for these settings could be improved and will submit an (untested) string change PR.

Sophist-UK, 2023-05-15:
I have just tested this again and it does appear already to snap as I am suggesting. Apologies.

closed 2023-05-15 by Sophist-UK