meerk40t/meerk40t #1900: [Bug]: Crosshatch burns outside of stroke on low dpi

dorsi90, 2023-05-15:
### Summary Description

When make a raster crosshatch with low dpi like 50dpi it burns outside of stroke line.
[image]
[image]

### MeerK40t Version

0.8.4002

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-05-15:
Hm. It actually looks like the laser in question needs a turn delay-on or delay-off setting. You can look carefully and it looks like every other line actually does this and the other end of the line does it the other direction. The low DPI means it moves pretty far between lines but you can see one side over-does the line and the other side of the line will be overshot the next time. So it's only overshooting on one side. 

I mean please save a gcode (or whatever laser file output you're using), but looking at the image there's some chance this is hardware related like the laser is staying on a split second longer than it should when turning off. I'll do my due diligence here but some lasers like galvo lasers and newlydraw lasers will absolutely provide some timing controls here and there and it really looks like your laser is staying on a bit longer than it should.

`Save` the job in `Plan` tab of the `Laser-Control` , and I'll be happy to check that the file actually says to stop where it should say to stop 25 DPI might also produce a less crisp image, like it actually reduces the DPI such that there might be some slop there but I would expect would cause an issue across the board rather than every other row.

tatarize, 2023-05-16:
Yeah, that's entirely possible. To do that operation it's creating an extremely small image and simple rounding error on an image that size could possibly result in exceeding those positions. If it's an m2 nano, you could probably save out an .egv file and I could probably check the expected positioning for that.

For speeds your y-axis moves have to move that entire laser bar and the laser head, so because of the weight differences it can absolutely cause some issues, especially at near top speed. In theory you could use a hatch operation for this effect. Or you could cheat and just use a much larger filled image with a hatch already in it, so you get it to look like a hatch but then just raster the image horizontally. Basically raster a picture a picture of a hatch. The actual steps here should be about 40 units which should maybe be a bit long between the scanlines but should work. 

Checking the .egv file is probably worth it, and really thinking about it, it very well may be causing some antialiasing around the edges and exceeding the positions at a lower overall power. Since if you're moving 40 units vertically you should also move 40 units horizontally which is probably enough to overshoot the end positions there.

In fact, I'm sort of convinced that's the issue.

It's taking the number `25` and making it extremely small. Then applying a raster to that. Downsizing the image by 1/40 from the full sized image. This means each pixel is technically about 1/40th of an inch wide and high. The height is generally going to be ignored and just gives you a scanline each 1/40th of an inch. But, the width is also affected by this. Meaning chunksize (so to speak) is going to allow stopping only in increments of 1/40th of an inch. And some of pixels will overshoot that amount, even if only slightly. When they do this, the power level of that pixel should be roughly equal to the amount of light pixels that were in the original image. So you should get them all going to the same endpoints but at slightly different PPI for the last little amounts.

This is why you will get very light lines on the inside loops of the 2 and 5. You can see the vertical lines there but clearly that area shouldn't have been black... but it could have been roughly gray. Basically the resolution at 25 dpi gets so reduced that you're talking 20 pixels or so at the fattest part of the raster.

tatarize, 2023-05-16:
For comparison, here's just normal text `25` about 15mm tall. Set at 25 dpi.

[image]

Here it is again with the trace line at the 25 or so.

[image]

tatarize, 2023-05-16:
[image]

This is the same thing done as a hatch. 25dpi is a distance of 0.025in so I typed that into the hatch distances. And did 2 one at 0.025 / 0deg and another at 0.025 / 90deg.

This won't invoke raster mode since it's purely a regular movement. So you wouldn't manage 500mm/s maybe 50 or so max. But, it would certainly work.

dorsi90, 2023-05-16:
@tatarize the problem with hatch operations is that it makes too much change of directions and do one path at time.
In raster mode it make scanline from top to bottom of all path at one time and also if is only one path it makes from top to bottom of entire path in one pass without change direction on each intersection that is good for k40 due of it's y axis.
So raster mode is perfect for doing this operation but instead DPI that change the resolution of the path i need something like line height (like in moshidraw software) in steps or mm that basically skips steps on y axis when engrave in horizontal direction and on x axis when engrave in vertical direction.
Is there any function in meerk40t that can do this?
this is the egv file, i added .txt extension for upload to github just remove it in case
[test.egv.txt](https://github.com/meerk40t/meerk40t/files/11486679/test.egv.txt)

tatarize, 2023-05-16:
https://github.com/meerk40t/meerk40t/assets/3302478/3eaa0cda-2d59-4c68-9015-11ee517b418a

The direction changes are that it's doing normal moves and can minimize distances there. You could turn off optimizations.

---

There isn't an easy method to do this as a raster and clearly a lot of reasons you might well want to do this as a raster. Hatch isn't a full solution obviously, and neither is raster. There is a function to maybe permit this to be done with rasters but it's not a function that is basically ever used outside of `ruidacontrol`. Basically there's some rastering in `ruidacontrol` that isn't specifically flagged as rastering so it needs to be detected and treated as such. Which means that if it sees a bunch of flat lines it will classify that as a raster and raster it (otherwise the rastering would be rather dreadful), though actually that caps out at 15 units moved vertically and you're at 40 so that metric wouldn't really apply. It might also be possible to avoid the high resample rate along the horizontal axis. So if I shrunk the data to like 15 pixels tall (the number of scanlines, but kept it like 600 pixels wide). In theory you sort of want to set horizontal DPI to 25 and vertical DPI to 1000 and vice versa. So rather than blend across horizontal pixels it does a hard-cutoff where it needs to do a hard-cutoff.

dorsi90, 2023-05-16:
@tatarize thank's for you video. I had already tried to disable optimization but only in the panel where there is the start button and then tried to simulate but only now i see that disabling the optimization in this panel don't disable automatically the optimization in the simulator so the simulation was altered and does not reflect the real job.
Later i try it and i think in my case it works very well. After the test i close this issue.
Thank you very much for your time

closed 2024-10-04 by jpirnay