meerk40t/meerk40t #1908: [0.8.4] CPU on idle very high i.e. 3%-10% of an Intel 4th gen I7

Sophist-UK, 2023-05-18:
When I came back to my production MK PC I discovered MK running at > 10% CPU despite being idle.

It subsequently dropped to between 0 and 3%.

[image]

The console `timers` command showed no timers active.

I am also unclear why there are two executables running.

tatarize, 2023-05-18:
1 is for bootstapping.

pyinstaller/pyinstaller#3173

https://pyinstaller.org/en/stable/advanced-topics.html#the-bootstrap-process-in-detail

The main thread I think cycles through a lot of checks to see if there's new things to do. And maybe that's doing the couple percent of cpu. Keep in mind, because of the GIL that 10% is actually 10%/25% (assuming my old 4 core machine). Since it can only max out one CPU.

tatarize, 2023-05-18:
As for the rest, there are two threads running at any particular time. These are `run` and `mainthread`

Run is the scheduler for the kernel:
```python
        self.state = "active"
        while self.state != "end":
            time.sleep(0.005)  # 200 ticks a second.
            while self.state == "pause":
                # The scheduler is paused.
                time.sleep(0.1)
            if self.state == "terminate":
                break
            self.schedule_run(
                self.scheduler_handles_default_thread_jobs,
                self.scheduler_handles_main_thread_jobs,
            )
        self.state = "end"
```

It's entirely just checking if anything is scheduled to run.

In the default case there will be around 6 jobs all of which are flagged to run in main, and will not be run. This could probably be avoided if the job and jobs flagged main were kept in different pools since these should not need to swap a job from gui-main to regular. And this whole thing could also use some threading.Condition flags to avoid running the scheduler at all if there's nothing in there. 

The regular jobs are run in a wx.Timer which then calls the scheduler for the jobs required to run in main, which is the signal listener and the updates for the 6 different scenes that exist by default (1 main scene, 5 hidden camera scenes). None of which require any updates.

tatarize, 2023-05-29:
#1935 Seeks to address this problem. Really 0.005 is pretty low and I don't really see any distinction with response times with 0.05 or so and it does seem to reduce idle activities by about a factor of 10. The signals are not intended to carry too much stuff by 20 per second times should be enough. I also made the scenes no longer try to refresh if they are absolutely not showing. There's a few hidden camera scenes that got weirdly placed in this sort of limbo.

tatarize, 2023-05-29:
I am unaware of anything in signals that should need faster response. They are intended to be overwrite repeated code last-state updates within the gui-thread. If this change breaks anything that thing was probably a bug anyway. Though I'm a bit torn on it since this basically means the smallest possible tick time is 1/20th of a second. Which basically means animations would be limited to that at the low-end.

It's likely possible to increase the next hit but the gui-thread is triggered with a wx.Timer and the regular processes runs in the scheduler thread (non-gui). And it's generally not very easy to modify that timer on the fly (it might be quite doable). The difference between waiting a minimum of 0.005 vs 0.05 seconds isn't much.

There's some methodologies for making schedulers a bit better. You basically change the sleep amount to be caused by the jobs themselves. So you basically store them in a heap and check the soonest value and wait until that one would hit. However we have a couple mechanisms where we can add conditionals that sort of mean it should trigger when those become true. You can tell the thread to `wait()` if there's no known future job, and notify it if there's a new job added. I guess I could also split the difference and make it like 25 ms (40 times a second). Since with the modifications it seems to hover at 0.

Sophist-UK, 2023-05-29:
As far as I am concerned the issue is CPU load when completely idle and GUI not showing. This doesn't mean reducing the tick when MK is active and reduce its responsiveness or spoiling the animations.

tatarize, 2023-05-29:
I don't know if it would spoil anything or affect the reactiveness. It might just be set too high in general. It only sends a signal here or there and sometimes would need to update the UI. But, sometimes it would do this back and forth a few times. Other times we'd never notice because if it takes longer than 0.005 seconds and it likely would it would skip the next round of ticks.

One of the changes I previously made was switching to a wx.Timer because previously I ran wx.RunLater() or whatever and those had a habit of building up where they couldn't complete the work in the given time period and thus would start getting a backlog of those things.

I fixed the backlog issue, but I didn't make the tick time such that all the work could possibly get done in fast enough.

Sophist-UK, 2023-05-30:
I have enough experience of this sort of python stuff on other open source projects to know that the details are DEFINITELY best left to you to decide what is best. (I just want MK to stop using CPU when idle, however that has to take second priority to literally ANY other functional need.)

closed 2024-10-04 by jpirnay