meerk40t/meerk40t #1921: [0.8.4] Image raster at v. low DPI burns alternate lines offset on Lhy but not in Simulation

Sophist-UK, 2023-05-19:
As testing the focus for a new head / lens, I rastered an image at v. low DPI (50). The raster lines do not overlap at this DPI so it is easy to see what is happening.

Here is the simulation:
[image]

Here is an image of the actual burn:
[image]

It isn't consistently the case, but it appears that lines are sometimes longer in the direction of the head movement. Most notable on the vertical lines.

I am not sure whether or not this is the same as #1900 (i.e. this is not a crosshatch burn).

P.S. It might be an improvement i.e. be more realistic if the simulation picture used a fixed physical height for raster rows when DPI is (say) below 150.

Sophist-UK, 2023-05-19:
As a comparison, here is the same image burned with uni-direction set:
[image]

As you can see, the difference in quality is noticeable.

tatarize, 2023-05-19:
Though I actually expect focus to be such that you more-or-less can see the overlap until 100 DPI or so. 
Even if it did convert to individual lines I would still generally overlap them in simulation there. However, this is an image and remains an image the entire time. The RasterCut actually keeps an image inside a bit of cutcode.

I would have previously agreed that maybe we should convert this over to just a bunch of lines, but newlydraw, for example has a mechanism whereby it converts the data during writing to basically a bitmap. You give it a bitdepth, step-amount, and then a stream of binary and it cuts that in a horizontal or vertical cut. I don't know if this works as nicely if it's already lines.

But, mostly lines are too slow there, and it might need some additional work converting that to geomstr data in 0.9.x to see if it can't be made fast enough to actually render as cuts instead of as a proper raster.

tatarize, 2023-05-20:
Tiger noted you may be highlighting an off-by-one error. Since you're right that every other pixel seems to go further. This was/is also the case at 1900. And it would be fairly easy to do. Since I'm treating pixels as discrete locations and more properly they should be treated as the inter-pixel locations between the pixels.

If we have a grid:
[image]

We would render that (in step 1) as UbDbUbTbDbUb. Since we go by pixels, however it's pretty easy to see that this could more correctly be seen as our location begins at begins at the pixel edge or even intersections. And our units are actually travels from pixel intersections to other pixel intersections. So our movement if our position is seen as the position of the pixel minus half-pixel in x and y. (or at least half pixel in x). Our movement is 2-off, 2on. 2off. Switch directions. 2-off 2-on, 2off. Which is what we had before.

Maybe this understanding is right but the actual implementation is wrong. Treating pixels rather than left-side of pixels as the full integer space. We'd actually expect there to be no pixel for the rightmost position. So when we ask for the right-most-filled pixel on the second row, the answer should actually be 3, located between green pixels (more or less) so width of 6 pixels, should have 7 different locations (since it can't move down within the raster space). And...

Maybe this system is wrong. We need 7 locations for a width of 6, and when moving right, our position is the left edge of the pixel and moving left our position is the right edge of the pixel. And maybe fix these our positions are going to be off by one pixel if we're going to the left.

There is quite evidently a slight bias of one pixel going bidirectionally and only every other line.

Sophist-UK, 2023-05-20:
> Though I actually expect focus to be such that you more-or-less can see the overlap until 100 DPI or so. Even if it did convert to individual lines I would still generally overlap them in simulation [...]

If it were me, I would try converting low DPI images back to (say) 1000DPI for display and then zero out pixel rows or columns which would be blank. This would then display much faster than considering rows or columns to be lines.

Sophist-UK, 2024-10-04:
This is a definite and reproduceable bug and it has not AFAIAA been fixed. Closing this is CRASS.

jpirnay, 2024-10-05:
Let me be clear: any issues for no longer supported versions have no chance of being resolved - so if you think this is still an issue in the current development branch then we would all be grateful if you can redo your test with the latest versions. Happy to pick this up at that point in time.

Sophist-UK, 2024-10-05:
This is almost certainly still an issue. It is not like 0.9 is a completely different code base from 0.8 (unlike 0.7 which was substantially different to 0.6).

However I will try to reproduce it.

Sophist-UK, 2024-10-30:
@jpirnay YES - this is still an issue (in 0.9.5b8 - I haven't yet switched to 0.9.5 full)...
[image]

[image]

I simply created 2 circles, merged them and set a fill and set Raster to 20DPI.

Sophist-UK, 2024-10-30:
And apologies for the poor resolution of the photo of the actual burn, but you can clearly see the 1/20" offset of alternate raster lines.

jpirnay, 2024-11-05:
I am still looking into it, but this is the current status:
- It's not a driver specific issue, it's a result of the implementation algorithm for raster (and image) operations.
- The operation calls a RasterPlotter instance that slices the image into line segments according to the desired burn direction 
- For a bidirectional operation the Overscan value is taken into consideration and added to the pixel calculation for the backswing. This leads to the observed overshoot.
- It's not a low dpi relevant topic the algorithm described above kicks in all the time.
- So if you set unidirectional then the overshoot disappears.
- Branch ``lhy_low_dpi`` has received an update to the simulation window that (optionally) breaks down the image into the line segments (a bit resource intensive hence no default), that shows the phenomenon
- So far this behaviour per se isn't necessarily a bug, the actual implementation needs to undergo a review though, as the effect does not fully disappear when overscan is set to zero.

Unidirectional:
[image]

Birdirectional:
[image]

Sophist-UK, 2024-11-05:
> * For a bidirectional operation the Overscan value is taken into consideration and added to the pixel calculation for the backswing. This leads to the observed overshoot.
> * So if you set unidirectional then the overshoot disappears.

Not necessarily - the overshoot could be on the unidirectional raster, but because it happens on every line you wouldn't be able to see it if this is the case.

closed 2024-11-07 by jpirnay