meerk40t/meerk40t #1923: [0.8.4] Special op Interrupt mis-processed

Sophist-UK, 2023-05-19:
I burned this test project twice.

The first time the interrupt happened before the second burn op, but the second time the second op burned whilst the interrupt popup was still showing!!

[image]

tatarize, 2023-05-26:
I attempted this twice both times worked. I'm going to need the actual file you used and maybe additional information as to how you ran things.... unless... nope. Yeah, I tried this 4 times nothing.

Sophist-UK, 2023-05-27:
There was no file. I wanted to test my nascent air assist (the AA in the labels) and so drew two rectangles, stuck an interrupt between them to allow me to turn the AA on or off and hit the big green button. First time it paused before the second op. Second time it paused AFTER the second op. There were literally no other actions between the two presses of the BGB.

My best guess is that the code which creates separate groups when you have merge ops set and a special op in between didn't work correctly.

tatarize, 2023-05-27:
It can't merge across an operation like that. I'm thinking maybe it got double placed in operations or something, and so it was waiting for the engrave but did the cut-engrave together or something. I can't duplicate it, and it wouldn't work one time then fail the next as it would optimize it identically.

Sophist-UK, 2023-05-27:
Yes - it was a weird intermittent issue which has only happened once. Sol incredibly difficult to diagnose.

closed 2024-10-04 by jpirnay