meerk40t/meerk40t #1936: [0.8.4002] Minor: Delete of Op (with Del key) deletes contents but not ops

Sophist-UK, 2023-05-29:


tiger12506, 2023-05-29:
I believe that if you hit delete on it a second time it will then delete the op.
It's my understanding that this is the desired behavior. Not sure I personally agree that that's the way it should be.

Sophist-UK, 2023-05-29:
Well, perhaps some people desire this behaviour, but clearly at least one person (me) doesn't. But that doesn't for a second mean I am right - just that there are differences of opinion.

closed 2024-10-04 by jpirnay

Sophist-UK, 2024-10-04:
This was definitely not completed as I had this issue today in the latest 0.9.5 beta.

As I said above, II (and apparently Tiger) think that the Delete key should operate consistently across all element types. But that is a personal preference and other opinions are equally valid - except that no one has actually expressed an alternative opinion.

jpirnay, 2024-10-04:
It's not a bug it's a feature: have a look at the console output when you try to do that - it is using the ``tree selected delete``  command, that will tell you that it deliberately constrained the removal operation and offers you an alternative, too: ``tree selected remove`` that will delete the operation  and its content.
So i you are sure that you want that behaviour, then redefine the del-key

Sophist-UK, 2024-10-04:
That is a how, not a why. The delete key could do a `tree selected remove` instead. It was a choice to use `tree selected delete` instead.

jpirnay, 2024-10-05:
No it shouldn't - you opt for a dangerous operation as a default

Sophist-UK, 2024-10-05:
Hardly a dangerous operation (deleting the Operation that you explicitly selected and then hit the delete key) especially when you can always use Undo.

Consistent UI actions are actually pretty important for creating a good experience for a user.