meerk40t/meerk40t #1952: [0.8.5b2] Crash on right click reload file node when file has been renamed

Sophist-UK, 2023-06-10:
```
MeerK40t crash log. Version: 0.8.4002 on Windows: Python 3.7.9: AMD64 - wxPython: 4.1.1 msw (phoenix) wxWidgets 3.1.5
Traceback (most recent call last):
  File "meerk40t\gui\wxutils.py", line 156, in specific
  File "meerk40t\core\treeop.py", line 120, in inner
  File "meerk40t\core\elements\element_treeops.py", line 1931, in reload_file
  File "meerk40t\core\elements\elements.py", line 3004, in load
  File "meerk40t\extra\inkscape.py", line 408, in check_for_features
FileNotFoundError: [Errno 2] No such file or directory: '\\\\Remote\\Data\\User\\Desktop\\Patterns\\Simple\\Simple.svg'
```

tatarize, 2023-06-13:
Fixed.

Sophist-UK, 2023-06-28:
In 0.8.5 it no longer crashes, but but the file node disappears without any explanation.

tatarize, 2023-06-28:
Where should it go? It's a file node pointing at nothing?

tiger12506, 2023-06-29:
My personal opinion.

If filename changed because we did a Save As... then file node should change to match.
If filename is incorrect because file went missing underneath on the filesystem... then Reload should handle gracefully. Perhaps this involves a dialog box that says "File is no longer available, do you wish to save your work?" And then present the Save As.. dialog if confirmed.

Sophist-UK, 2023-06-29:
Yes - that is the workflow I use - however sometimes I rename the file to include the material size it needs. And then when I try to reload it, the reload fails.

Of course, it would be great if Meerk40t had the prescience to know what had happened to the file and reload it anyway, but that is definitely not what I expect the solution to be. 😉

tatarize, 2023-07-13:
The only real way to do this seems like if there was a rename within meerk40t that would rename the file and would go ahead and update the file itself. Which seems a bit beyond the regular scope of the program but would at least work. And if you renamed it outside of meerk40t it would just lose it per usual. And would only serve this one very minor sort of event.

Sophist-UK, 2023-07-13:
As I said, I am **not** expecting MK to track file renames. Although technically Windows - and possibly other o/s - allow you to register a callback (whilst your executable is executing) to be told that files of interest have been renamed or deleted, this is overkill. But I do expect a message saying why the file I was expecting to be loaded wasn't.

closed 2023-07-13 by Sophist-UK