meerk40t/meerk40t #1966: [0.9.0b2] Issues

Sophist-UK, 2023-06-15:
The latest BETA release of 0.9.0 is Beta 2. This issue is for reporting issues with this specific version of MeerK40t.

Please do try this release, though as an early beta there is no guarantee that this is suitable for production use. Please do not report any further bugs with previous beta releases or indeed any releases other than 0.9.0 Beta 2.

This issue report wants to cover identified bugs / issues with this specific version of Meerk40t to help resolve them. Please be specific with your report and if feasible attach relevant infomation (screenshots, the file you used etc.) to it, so that the developer team can try to recreate the issue.

If you want to provide feedback on other subjects close to your heart please use appropriate channels:

General discussion: https://github.com/meerk40t/meerk40t/discussions
Feature requests: Please open a seperate topic in this channel: https://github.com/meerk40t/meerk40t/issues/new?assignees=tatarize&labels=Type%3A+Enhancement&template=suggestion.yml&title=%5BSuggestion%5D%3A+

tatarize, 2023-06-16:
Mostly and importantly:

* Node-Edit does not work. This prevents it from emerging from beta status.
* Op-Hatch is a bit weak currently. Compare this with `effect_hatch` and doing `element* geometry hatch node` which both work really well. The problem is op-hatch has some, extra features like performing 1000x hatches at small angle turns though done with pens and sort of hidden. That makes the implementation a bit harder.
* The .exe lacks libusb support. It can talk with the WinDll, but not with the libusb driver. This failure may also break Galvo support which currently relies on that functionality. This isn't a problem with the code, but with the build.
* Still adding a few standalone features, eg. #1967 (Reading lightburn files directly).
* Probably needs some addition UI tweaks. #1968

[image]

The idea here is that we can quickly do 1 (Draw something), 2 (Set settings in Operations), 3. (Start the Laser running).
IncredibleDeath in the discord support seemed to think these were improvements and recommended the ribbon-bar be made slicker.

tatarize, 2023-06-16:
## Ruidacontrol should be taken out of beta. 
IncredibleDeath went through most of the versions available and not, and tested them with ruidacontrol and grblcontrol. He provided an easy test example of failure that highlighted the issue. 

### Corrected Problems.
* Fixes in 0.8.x for power level not being respected.
* Added a much more complex view of the .rd files in the blob-properties (this is needed to make sure files are files and probably needs additional analysis to get some file send and play functionality more fully working).
* Reduced the h_switch_g excess_delta value from 15 to 5. If you were in a raster and the raster was a bit bigger than we had expected, we would just tag on that extra amount without exiting raster mode. It turns out 15 still makes a highly audible click. In most cases we'd have a DPI that causes a step value (1000 / DPI) of between two numbers e.g. 4 and 5. So our g value is set to `G004` and we merely adjust upwards by 1 in the case that we need the 5 value. This will in regular use be an excess_delta of 1 and in the case of a missing scanline (e.g. entirely white) we'd move upwards by another 4-5 delta).

### Remaining problems.
Basically the problems are in series 7 things are clicky this is because of the use of NSE-rastering which was deprecated because it was inherently clicky. But, version in series 8, an example test image...

[image]

https://github.com/meerk40t/meerk40t/assets/3302478/fc5e1fc8-39d5-4be3-bf94-5d5b97c94f61

Suffered raster as vector error. This is because when lightburn ran it the `tes` and the `t` were treated as two different filled objects. And it would raster the `t` upwards. Then move to the `tes` and raster that upwards too (or some arrangement of that), which meant that it moved in all four directions by amounts greater than 15 in all directions. This forced the series of points to get flagged as a `vector` rather than `raster`. Contextually the series of points are given without actually anything flagging what is a cut or a scan, and we can't expect to have that information given to us. So we have a series of points with our laser on/off/power levels.

### Workaround
The fix that finally worked for him is a bit of a hack:
https://github.com/meerk40t/meerk40t/pull/1962/commits/9d2d8b0c3ab0f37a3f92a5b0cb812c0cfd7a52a5

This simply forced everything run faster than 80mm/s to count as a horizontal raster. This basically rules out vertical rastering and doesn't correctly and fully solve the problem. But, it does solve 95% of it for any regular user.

### Fix
The fix is probably to actually flag the different sections correctly. This means, rather than each series of points at the same speed, we may need to see if there are subsections within that section which are functionally a raster. Rather than a heuristic saying the entire series is or is not a raster. Or we could simply execute two different heuristics to determine if the vast majority of points are a raster at angle 0 or a raster at angle 90° and flag the entire section in that manner (if it's faster than the 80mm/s cutoff or so), and flag them all. Giant movements against the direction of the raster will force the raster out of raster mode. The workaround did actually raster perfectly fine for him.

### Finishing
Many of these problems would be fixed in `grblcontrol` too since there is sharing of code, but there were some other issues with that and it would need independent testing.

Sophist-UK, 2023-06-16:
> IncredibleDeath in the discord support seemed to think these were improvements and recommended the ribbon-bar be made slicker.

As we have discussed several times, good UX is difficult, and I would caution against using input from a single user (whether it be IncredibleDeath or me or anyone else) without validating with other experienced users whether they would agree with whether A) The functionality is useful enough in general to be worth writing and prioritising to developing now, and B) whether the UX design is a good one.

I am all for continually including new functionality for additional controllers, for additional emulation and for functionality that e.g. Lightburn also has - that is how this already-fantastic software will become even greater - however I do believe that the priorities should be:

1. Cleaning up the existing code (e.g. with Type hinting that helps detect bugs) and writing tests to confirm correct functioning (e.g. the various recent m2 nano issues that crept in) should be top priority, even though this stuff is boring to code.
2. Bringing back functionality from 0.7 that has got missed. Resolving some of the other backlog of issues - again not sexy, but will make users happy.
4. Continuing to improve the UI for existing functionality and help functionality to make it more informative to new users and intuitive with contextual help - so e.g. making all context commands work on multiple selections, resequencing and grouping context commands and spacers, improving the console help command, providing more status-bar help mouse down pre-execution and for hovers over buttons etc. and by e.g. including panel help buttons that go to specific web pages, 
6. Removing requirements on other software e.g. installing whisperer to get LibUSB drivers. (Can we not deliver some simple functionality to download and install the drivers if someone needs them installed?) Can we create functionality to avoid Inkscape restrictions or write Inkscape plugins to do better cleanup that you can't do with Inkscape native?
7. Making RUIDA / GRBL emulation rock solid for users who want to use Lightburn.
5. Perhaps having some online video / demo sessions with real users to brainstorm UX improvements. Those of us who use MK a lot for real work can probably give a lot of good feedback on what is repetitive and / or time consuming in MK - and these are not necessarily what you might think. Things that come immediately to mind (not all of which are necessarily ones I would use myself): 
    * hard stop predictions (i.e. the head will hit the hard stops because e.g. design is too big or rasters are too near the edge), 
    * enabling / disabling group or element nodes, 
    * clever repetition of designs i.e. rectangular step repeat, auto nesting in both rectangular material and (if you have a bed-cam) any shape material - or any other functionality that a user would have to do manually which are time consuming
    * clever design functionality that no other software has such as: geometric hatching (e.g. spirals), pseudo 3d engraving (hatch lines following contours), translate 3d models to (plywood) layers, 3d carving (heavy burn to a depth) etc.
    * hardware maintenance assistants - which will make alignment, focusing etc. easier. Like the calibration patterns but cleverer

    These would be sexy functions and exciting to code.

tatarize, 2023-06-17:
Squeaky wheel gets the grease. Nobody else is suggesting better UI and it's fairly easy to switch around. I just mostly need to be able to modify what's shown in a toolbar fairly dynamically.

1. I don't think there's any test to really automate for that. We have a sort of tiger test protocol. 
`Before a full release, at least one of us needs to confirm pause/unpause, and estop functionality on the big 3 laser platforms. Additionally, a successful vector and raster burn. That's my bare minimum condition I'm imposing for good measure.` -- And maybe we should actually try that stuff with the compiled versions, but I'm not sure how exactly that would be done since we'd need to kickoff the builds in isolation or something.
2. I am planning on adding Passthrough again, likely by switching nodes over to `image raw` and `effect wizard` or whatever to create the raster wizard'd scripts. Currently they are sort of weird but I think I kind of figured out a workable methodology when doing `effect hatch` work. This could also, possibly include some methods for dividing and image up manually. Though that's sort of speculative.
3. Somehow I'm not sure throwing more info at them makes for a better UI. Though I could see reducing the number of buttons to somewhat fit to one screen rather than 4 pages in the ribbonbar.
4. Nobody will need LibUSB drivers. They aren't needed. If you have them you have them, but it works without them.
5. The emulation should work, is nearly there and I do intend to make sure it's decidedly quite solid. And I'll lean into loading `.lbrn` files as some parts of that.
6. Yeah, I do need significantly more user feedback. 
    * This is actually intended, there's code in `placements` where you can place a job at multiple different locations etc. And it checks to see that it knows the size of the job, the job size and bounds are known during the pre-validate step.
    * You can already disable some nodes, though generally the op-nodes and you can drag the other node types into regmarks. In theory there's some ideas to let you swap things into basically a group in regmark called Layer1 or something. Then you'd just swap the data in and out of elements node. So all layers pulls all groups to the elements. Hit a button send that group to regmarks or something like that.
    * You can basically do that with Placements. You can set an set of automatic run at this location nodes. And it will run at the location of all of those placements. This includes the directional of the placement if required, and which corner the placement gets placed at (or the center).
    * Ezcad2 has spiral hatches. I've had some reasonable ideas with regard to other more indepth operations. Like, z-depth rasters where you can divide a z-depth image into subimages. But, there was always a difficulty visualizing this. Since you'd sort of have pages. The same is true for something like hatches that increments angles. You step forward and get a new hatch where you're hatching at a slightly different angle you would not see this in most visualizations because the previous hatches would block your view. So it makes viewing and simulating that stuff difficult. But, basically you'd run this like an `effect wizard` with `effect zdepth` and it would contain a series of images, rather than like 10 images that you'd do in passes that are annoyingly overlapped and if you move one it gets to be an annoying. The 3d models to plywood stuff is actually a slicer. Though in this case you'd kind of want to keep the solid stuff and drop the non-solid things to allow gluing them together. It sort of feels out of the scope and there's a lot of really good slicers for 3d printers.
    * I used to have a sort of alignment helper thing. It didn't seem like it ever helped. I didn't even use it when I needed to align my laser.

Sophist-UK, 2023-06-17:
1. Yes of course there is. You can input an SVG element (or a CutObject) and get the commands sent to the m2 nano and check they are as expected.
2. And raster optimisation.
3. Good UIs are hard - and however hard it is to make a good UI for expert users or alternatively for new users, making one that is great for both types is even harder. BUT it is possible, but only by harnessing a synergy of a TEAM of REAL USERS (beginners and experts) talking things through. But IMO , in essence a good expert GUI (and MK isn't bad) can be turned into a good  beginners GUI with five methods that don't impact experts:
     1. Dynamic immediate status bar help; 
     2. Tooltips (though you have to be careful that they don't get annoying to experts 0 so maybe have an expert switch for these)
     3. Right click context help at the bottom of every right click menu
     4. Other buttons / tabs to get context help on e.g. panes, toolbars
     5. Wizards to take new users step by step through things that experts don't need to invoke.
4. Ok. 😄  I will remove mine at some point.
5. Great.
6. There is a BIG difference between occasional user feedback (often only expert users) and actively **soliciting a group of users** to give you a consensus view. The list of things I added were only to give you a sense of what is possible - remember I explicitly advised **against** doing things on the input of just one user.

But I think you need to decide what your personal goals are for MK. If you just want to create a tool for experts that can do cool stuff, that is great - I am good with that because I consider myself an expert user, so this isn't me being critical or preachy. But if you want to create the worlds most amazing software that can be used by a range of users including beginners, and which is the software everyone mentions first instead of Whisperer or Lightburn, then you need to accept that this needs a slightly different mind set with a significant priority on quality and on usability.

tatarize, 2023-06-19:
1. Yes. There's some checks to see if the data produced by the drivers is the same as it previous was. But, that's not really the issues we had with the m2 Nano. The bytecode was fine, the sending protocol was not.
2. I've not really thought of a very easy way to restore the raster optimization code. There's check the same things your code had checked but I feel like there's probably a better solution if geomstr is working and faster to do a much more proper raster-like optimization based on the given path information.
3. So I've been toying with replacing the ribbonbar with my own custom-like toolbar. 
[image]
Which is not as helpful as doing a good job, focusing on function currently. But, somehow I hope that more basic control will let me start multiple tool bars that are all dynamic and maybe make things more easily fixed. That or increase the number of icons on the screen, I guess.
4. Not really a point to removing them either. But, certainly no reason to provide them during install (also it would require an install).

Hardware is a bit out of scope for the project. Seems like at most a video, and really alignment is one of those things that you need to understand to understand or you'll be lost in the weeds for months.

Sophist-UK, 2023-06-19:
3. My personal approach is to reuse as much stuff already written by others as possible (providing that it has longevity and is compatible and ...) rather than try to reinvent the wheel. 

The problem with wxPython is that it is a Python interface to a native C library, so even though there are a few extensions to wxWidgets, they probably are not usable by wxPython.

In addition, I suspect that recent (i.e. last 10 years) history is littered with examples of technical people who disliked the way a GUI widget worked, and decided to write their own and then got swallowed up with making it work and supporting it afterwards. 

I know it would be a big deal to switch to an alternative graphics library, but my advice would be try to find a library that is python-centric and has all the functions you now want, and then think about whether it would make sense to make v0.10 to switch over. In the end it might be just as little effort to switch over as to start writing your own low-level widgets.

closed 2024-10-04 by jpirnay