meerk40t/meerk40t #2018: [0.8.4] Warning sign on Op node not cleared...

Sophist-UK, 2023-08-02:
I had deliberately deleted some Op leaf nodes for a file. This correctly gave me a warning shield on the Operations node.

I then imported a new file, and MK added the elements to the Ops - then I removed the old file.

At this point, all elements were in Ops, however the warning shield was **not** cleared from the Operations node.

jpirnay, 2023-08-02:
1. Fixed for 0.8 in [4a179b2](https://github.com/meerk40t/meerk40t/commit/4a179b2a3c6fe822cc74fbd32bf569184e7a7d49)
2. Fixed for 0.9 in [7db7ed9](https://github.com/meerk40t/meerk40t/pull/2019/commits/7db7ed95cdc903a16e17bea990cef10a3c6564b3)

Sophist-UK, 2023-08-02:
Thx Jens - that was very fast.

closed 2023-08-02 by Sophist-UK