meerk40t/meerk40t #2022: [0.9.0b4] New `Ribbon Bar`: first impressions

Sophist-UK, 2023-08-03:
My first impression of the new Ribbon Bar is that I hate it, but I immediately realised that it was not the functionality that I hated but the superficial things like colour and font and so forth which are relatively easily fixed.

Please note: these comments represent a subjective view of personal design likes/dislikes and as such should only be acted upon if there is some general consensus amongst the devs.

### Font
This currently uses a Serif font whereas the whole of the rest of MK uses a Sans Serif font. This really clashes IMO.

### Colour
A. The lack of colour background fill was the other thing that jars in a major way. I am not particularly wedded to the blue background in previous versions, but I feel that using a different background colour to the other panes helps to highlight that this is a more important pane.

B,. Highlight colours are IMO too dark - they are overpowering and make it less easy to read the text. I am not wedded to the old colour scheme particularly, but the way the colours changed for active tab or as you moused over the sub-groups and buttons was much better in previous versions.

### Group boxes
The curved corners are different to all other panes, and this also jars. I would suggest losing these curved corners.

Additionally, it is difficult to know what it will look like without curved corners, but it might look as good/better using a single box with vertical bars separating the sub-groups.

###  Truncated text
"Naviga..." vs. "Navigation", "Control..." vs. "Controller". This may be fixed by a change of font as SS fonts are often slightly narrower than S fonts. Alternatively, shrink the font slightly for all button names until they all fit.

### Bottom Margin
We have lost the sub-group titles at the bottom (which is not an issue for me) however that means we now have a bottom margin which is bigger than it needs to be. Can we reduce the minimum height of this pane?

### Tabs
A. Can we centre the tab titles in the tabs?

B.  We have lost the design tab - because it has been replaced with a Tools pane. I think the Design tab (or a Tools tab) should still be there, with the Tools pane as an option.

C. Gap between tabs perhaps should be narrower - or alternatively mirror the functionality of previous versions where the tab outline only appears when active (which makes it feel more like a hybrid tabs/menu than pure tabs).

### Tools pane
I like this concept, but there are **several** issues with this new pane.

A. Minimum pane width is too narrow with the vertical layout it needs to be set so that the icons are always shown.

B. Vertical-only layout is way too inflexible - the layout needs to be adaptive to the width of the panel it is in so that you can have it in a horizontal panel or together with other panes in a wide vertical panel. **This is the only thing stopping me from moving to 0.9 for permanent production use.**

C. Tooltips do not appear when you mouseover a disabled tool (e.g. if you have nothing selected)- this is less important (so if it is difficult to achieve forget it), but I think it would be helpful to still have a tooltip over disabled items (perhaps with additional text explaining that you need to select items to enable it).

### Functionality
Wordlist button has lost its dropdown submenu.

### Modify tab
Icons are fuzzy/illegible. Slightly large or thicker-lined icons would be beneficial.

P.S. I did a first burn in this beta and it worked fine. I look forward to experimenting with the improvements once the Tools pane layout is improved.

tatarize, 2023-08-03:
>My first impression of the new Ribbon Bar is that I hate it, but I immediately realised that it was not the functionality that I hated but the superficial things like colour and font and so forth which are relatively easily fixed.

And the stuff that it fixes were really hard to deal with and not superficially fixed. They were, in fact, pretty much terrible solutions. 

> Please note: these comments represent a subjective view of personal design likes/dislikes and as such should only be acted upon if there is some general consensus amongst the devs.

Nah, frankly all the devs, including myself aren't that great at design stuff. And this was more whipped up to fix the issues where the older ribbonbar was kind of stuck into the given methods laid out in the ribbonbar class.

> A. The lack of colour background fill was the other thing that jars in a major way. I am not particularly wedded to the blue background in previous versions, but I feel that using a different background colour to the other panes helps to highlight that this is a more important pane.

This was sort of due to the colorlessness of dark mode ribbonbar. Since it was the color of the background stuff, and this sort of copied that.

> B,. Highlight colours are IMO too dark - they are overpowering and make it less easy to read the text. I am not wedded to the old colour scheme particularly, but the way the colours changed for active tab or as you moused over the sub-groups and buttons was much better in previous versions.

Yeah, this is unknown why this color change happens it could be that I failed to understand the art or it was doing a much more pronounced gradient and thus was *much* dimmer.

>The curved corners are different to all other panes, and this also jars. I would suggest losing these curved corners.

It was copied, moreorless, from the original ribbonbar.

>We have lost the sub-group titles at the bottom (which is not an issue for me) however that means we now have a bottom margin which is bigger than it needs to be. Can we reduce the minimum height of this pane?

>B. We have lost the design tab - because it has been replaced with a Tools pane. I think the Design tab (or a Tools tab) should still be there, with the Tools pane as an option.

Tools sort of got moved mostly to the tools bar at the edge. The hope was to reduce the total number of icons there to something much more reasonable.

> B. Vertical-only layout is way too inflexible - the layout needs to be adaptive to the width of the panel it is in so that you can have it in a horizontal panel or together with other panes in a wide vertical panel. This is the only thing stopping me from moving to 0.9 for permanent production use.

Not sure if it needs a Pane Reset or the latest version of the code, but it works in either orientation.

> C. Tooltips do not appear when you mouseover a disabled tool (e.g. if you have nothing selected)- this is less important (so if it is difficult to achieve forget it)

This was actually done by default and we added code to suppress them in that case since they were disabled and the original ribbonbar didn't show such things.

> Wordlist button has lost its dropdown submenu.

There might be 2 of them.

> Icons are fuzzy/illegible. Slightly large or thicker-lined icons would be beneficial.

I think the icon size is determined by the space available. 

---

Bit of an older list of differences I noted between the old and new bars. 

As far as the older version goes. We also had:
* Names of panels on the panels at the bottom that did sort of annoying highlighting when moused over.
* The mouse over color was a tint of yellow gradient, not like total blue.
* The tab mouse over was only the top in lightish blue and bottom remained the same (might have been supposed to be a gradient)
* If the width was not wide enough to support all the icons the panel with the most item would double up the last 2 items, until no two items could do this. When all items were done, the next longest was applied in the same manner.
* If the width was still not wide enough, the panel would be minimized into a single dropdown of minimized mode.
* Position of the button were equally spaced along the height for the excess height that was available.
* Pair stacked buttons had their labels on the right.
* When the height was not-sufficient, all the panels would minimize making the whole thing sort of look horrible.
* The  path around the page and tab didn't have a boundary, it went around the table and then around all the panels all the way to the edge. It was not two distinct rectangles.
* Text label had real type-setting rather than each space. It would break text in a greedy fashion with regard to width.

Sophist-UK, 2023-08-03:
Thanks for considering my opinions.
-------------
@jpirnay: 

I have not tried resizing the ribbon bar except for trying to see what happens if I reduce the width when on minimum height - and that is what my comments about icon/font sizing are based on. The icon autoscaling looks very good to me based on this very limited testing - my comments on ribbon bar sizing were focused only on text autosizing. 

However the screenshots above of Tools pane resizing was not the experience I had. I turned on the Tools pane for the first time, and it appeared in its own vertical panel. I was able to reduce the width of that panel to much much less than the width of  a single icon. I then docked it (by accident) on another vertical panel which was significantly wider and it didn't look like the pictures you posted - from memory the frames were the correct width but the icons were not laid out as shown in your screenshots.

And yes - the purpose of a beta program is to get feedback!! Hence my feedback. (Though I also give feedback on full releases.)

---------------
@tatarize: 

> And the stuff that it fixes were really hard to deal with and not superficially fixed. They were, in fact, pretty much terrible solutions.

You need to do what you need to do for functional purposes and then make the look as good as fit as you can as a secondary priority.

> > The curved corners are different to all other panes, and this also jars. I would suggest losing these curved corners.
> 
> It was copied, moreorless, from the original ribbonbar.

This may of course be a Windows vs. Mac vs. Linux thing. On Windows the frames have square corners (or possibly a tiny radius curve) but definitely not like this.

But IMO home grown widgets need (if possible) to match the style of the standard widgets on each operating system - which may mean o/s specific code on the home grown widgets to make it match for each o/s.

> Nah, frankly all the devs, including myself aren't that great at design stuff. And this was more whipped up to fix the issues where the older ribbonbar was kind of stuck into the given methods laid out in the ribbonbar class.

As I have said previous UI stuff is hard in an absolute sense, even for UI specialists, and it is as much of an art as a science too. And my experience in IT for 40 years suggests that most (if not all) developers have difficulties creating good UI/UX. So don't feel bad about this!!!!

> Tools sort of got moved mostly to the tools bar at the edge. The hope was to reduce the total number of icons there to something much more reasonable.

As I say, I like the idea of the Tools pane very much. I would suggest that the default for the Tools pane is ON rather than OFF. The reason I missed the Design tab is because the Tools pane was not showing by default - but if screen real-estate is an issue, then the user might still prefer a tools ribbon bar. (As an aside, I think that having screen layout profiles might be a nice idea - we can populate the default list with a few examples, and users can then save and load (and export/import) their own.

> This was actually done by default and we added code to suppress them in that case since they were disabled and the original ribbonbar didn't show such things.

Well - having found the Design tab missing and made the Tools panel appear, I mouseovered each icon to see what it was and got no tooltip for those that were disabled, and without knowing what they did I had no idea how to enable them to get a tooltip in order to find out what they did. (I guessed because I am an experienced user - but others just starting with MK and trying to work out how the UI operates may just find it confusing.)

> > Icons are fuzzy/illegible. Slightly large or thicker-lined icons would be beneficial.
> 
> I think the icon size is determined by the space available.

Yes - I already commented that I like the autosize - but at all the various icon sizes, the icons need to be clear rather than fuzzy. You may need to use a paint program like GIMP to tweak the icons so that they look good at each size.

> * Names of panels on the panels at the bottom that did sort of annoying highlighting when moused over.

Actually, (on Windows at least) it was highlighting on mouseover of the panels themselves. And I find it a good UX because it helps you to understand what is happening in a very low key way. As I already said, I do **not** miss these sub-group names.

> * The mouse over color was a tint of yellow gradient, not like total blue.

Not yellow for me - on Windows (or Windows with the colour scheme I use). Part of standard widgets is that they adapt to O/S and to the O/S colour scheme chosen by the user, and home grown widgets really ought to do the same (though it is a PitA to have to program this).  

> * If the width was not wide enough to support all the icons the panel with the most item would double up the last 2 items, until no two items could do this. When all items were done, the next longest was applied in the same manner.

This old behaviour was OK, but I actually prefer the new autosizing behaviour anyway - so a good UI decision AFAIAC.

> * Text label had real type-setting rather than each space. It would break text in a greedy fashion with regard to width.

Yes - wrapping text onto two lines is OK but IMO truncating using elipses is not.

tatarize, 2023-08-04:
> Tools pane is ON rather than OFF. 

It should have been on. Really I think the issue is that the saved perspective is still there and valid and it didn't have a tools pane. I would think it would show up but it's entirely possible that it doesn't. The same is true for the ribbonbar not able to dock for the side panes, where that's actually just that the settings were saved in the `.perspective` in the settings and thus is sort of stuck that way. Does it for min and max for the ribbonpane too which is sometimes a problem.

>  I mouseovered each icon to see what it was and got no tooltip for those that were disabled, and without knowing what they did I had no idea how to enable them to get a tooltip in order to find out what they did.

Could do mouse over on disable since disabled is just art. But, I'm not sure. `(DISABLED) Blah Blah` maybe. Dunno.

>Part of standard widgets is that they adapt to O/S and to the O/S colour scheme chosen by the user,

Yeah, I used the previous art that @frogmaster set up for the routine and some of it was taken from the background of the users OS. 

There's a bunch of subtle stuff in the UI there, but the fact that we can have dynamic vertical tools matters a lot. Also, for very small screens it doesn't keep taking up most of the screen and is apparently usable.

There's also an issue with the individual node-editor tool. The older version before I broke it and lost most of the features, would make it's own toolbar, and I sort of wanted the ability to swap out the side toolbar there. Though I didn't yet, code that switch in, fix the node-editor to restore the features, or use it for that, it was (before this) impossible to do that.

jpirnay, 2023-08-08:
@Sophist-UK I understand you are using released versions for your tests only (and I get that), still the latest source does contain some adjustments to take the current feedback into consideration. (Some of it already in the latest beta, but not all). Maybe worth a look or two.

Sophist-UK, 2023-08-08:
I downloaded the new beta earlier but haven't had a chance to look at it yet. Whilst I intensely dislike the look of the new ribbon bar I can live with that for real use. But I need to see if the Tools pane is useable or not - I only had a quick initial impression last time and it didn't reconfigure itself well based on the panel size when I tried it, but it really needs more testing to see if it is liveable.

As for my existing ribbon bar critique, A) I don't expect you to change based only on the opinions of a single user, and B) such changes take time. So don't worry - I am not expecting perfection, not yet at least 😄 .

Sophist-UK, 2023-10-11:
What is the point of people like me giving early feedback on the usability, look and feel of BETA versions if you take no notice / take no action on the comments made in all the subsequent betas and release candidates.

I appreciate that functional issues need to take priority over form issues, but still there appears to be a complete lack of interest in whether MK has a good look and feel or not.

jpirnay, 2023-10-12:
Summarising the status of the different issues might help:

- [x] This currently uses a Serif font whereas the whole of the rest of MK uses a Sans Serif font. 
- [ ] Colour background fill too dark
- [ ] Highlight colors too dark
- [x] Round corners of group boxes _were reduced in size but have been there in  previous versions too_
- [x] Truncated text
- [x] Bottom Margin
- [x] Centre the tab titles in the tabs
- [x] We have lost the design tab. _We have added a toolbar designer in settings where you can add that again if you like_
- [x] Gap between tabs perhaps should be narrower
- [x] Minimum pane width is too narrow
- [x] Vertical-only layout is way too inflexible
- [x] Tooltips do not appear when you mouseover a disabled tool
- [x] Wordlist button has lost its dropdown submenu _see comment above_
- [ ] Icons are fuzzy/illegible. Slightly large or thicker-lined icons would be beneficial _This would require a major overhaul of the icon scheme (and the terms of use of relying on icons8)_

Sophist-UK, 2023-10-12:
* [ ] No colour background.

jpirnay, 2023-10-12:
I have opened a new issue helping us to focus on the remaining tasks - if there are no objections I would close this one tomorrow (the new issue is linking to this discussion for reference)

Sophist-UK, 2023-10-12:
Closing in favour of #2154

closed 2023-10-12 by Sophist-UK