meerk40t/meerk40t #2027: [Bug]: Settings not Saved / loaded

StefanX13, 2023-08-07:
### Summary Description

The Device Settings wont be saved / loaded correctly

### Additional Details

I have configured 2 devices one named K40x an one K40 rotary.
Since engrave one type of glas on regular basis I wanted to setup one config for "normal" operations and one for rotary, so i can just switch to the 2nd device without changing all the settings every time.

It saves both devices but not the rotary settings of the 2nd device.

I already tried to change the settings manually in c:\users\username\AppData\Local\Meerk40t\Meerk40t.cfg by changing the following:
`[lhystudios1]
registered_path = provider/device/lhystudios
name = LihuiyuDevice
extension = egv
label = K40_Rotary
board = M2
source = co2
user_scale_x = 0.9786
user_scale_y = 0.9786
flip_x = False
home_right = False
flip_y = False
home_bottom = False
swap_xy = False
autolock = True
plot_shift = True
strict = False
twitches = False
opt_rapid_between = True
opt_jog_minimum = 256
opt_jog_mode = 0
rapid_override = False
rapid_override_speed_x = 50.0
rapid_override_speed_y = 50.0
fix_speeds = False
rotary_active = True
rotary_scale_x = 1.0
rotary_scale_y = 0.386`

When i now start Meerk40t it loads the K40_Rotary device but not the rotary setting as in the cfg.

### MeerK40t Version

tested with 0.8.700 and 0.8.600

### MeerK40t Type

Executable

### Your Operating System

Windows

tatarize, 2023-08-07:
Hm. Seems like editing the settings should also work but really it should work anyway. What setting doesn't save the enable of the rotary board? In theory it's supposed to let you toggle the rotary on and off since the idea is that the device would be the same and it would either have or not have the rotary.

StefanX13, 2023-08-07:
I tested it a second ago, if i change the settings in meerk40t it saves the cfg according to my settings in meerk40t when i close meerk40t. But when i restart it, it doesnt load the rotary enabled as well as the rotary settings for y in my case 0.386.

Btw. its the same with the window settings. If i change them to annother layout it doesnt save it, and when i restart everything is in the default layout. JFYI

tatarize, 2023-08-07:
So the rotary_y value isn't reloading correctly?

StefanX13, 2023-08-07:
No it doesnt load the enabled as well as the value

tatarize, 2023-08-07:
Confirmed. That did not save.

tatarize, 2023-08-07:
Well, successfully saved but didn't reload that.

rotary_active = True
rotary_scale_x = 1.0
rotary_scale_y = 0.386

StefanX13, 2023-08-07:
Btw. while looking closer in the cfg i found this at line 101 and up:

```
[rotary/0]
index = 0
rotary_enabled = False
axis = 1
scale_x = 1.0
scale_y = 1.0
```

That seems not related to any device, the first device (K40x) is configured starting at line 188 and higher and the second (k40_rotary) at starting line 502

tatarize, 2023-08-07:
There was some debate as to where to put the rotary settings. Should they be a per-device setting or should you allow multiple rotaries that are not tied to the device. There's a bunch of other lesser issues with rotaries but the rotary scaling was supposed to work even if we hadn't dealt with stuff like split-area and non-axis replacing rotaries.

StefanX13, 2023-08-07:
Ah, ok. I tried to delete the first rotary related entries at line 101, but doesnt change anything. It saves all values but dont load them on restart.

tatarize, 2023-08-07:
Yeah, it wouldn't it seems like there's something forcing a reset of the settings between registration of the choices and execution.

`set -p lhystudios`

Shows the setting as reset even though I watched the choices load up correctly.

tatarize, 2023-08-07:
Okay, while the user_scale_x and user_scale_y settings would work, it initializes the device with default settings which are device settings and thus restore the rotary settings to default.

You could set the `user_scale_?` values right but the rotary stuff is getting reset, lemme fix that.

StefanX13, 2023-08-07:
Ok, thanks a lot!

PS: Dont want to open the debate again, but i see them in the per device settings so it could be a workflow to save "virtual devices" which store the rotary settings as i tried. The other solution would be a separate rotary setting which could be saved for e.g. different glases and could be loaded/chosen in the rotary tab. 
e.g.
Rotary Setting Glas 1
Rotary Setting Glas 2
a.s.o.

tatarize, 2023-08-07:
Hm. That is a really great point. I usually considered the chuck-type rotaries rather than the rollers which would probably want different settings per-tumbler-type rather than 1 per rotary device. 

It's a fine thing to open up. It's certainly not solved by any stretch. We just decided that we need to at least give as much rotary support as was available for 0.7.x

StefanX13, 2023-08-07:
Would love to have the chuck style one, but based on the price of this type of rotary, i suspect that most of the K40 Users have a roller type, but i can be wrong. Maybe i have a look for a printed chuck type or try to design my own one which can be 3d printed...
btw. love your work, at first meerk40t is a bit overwhelming, but after a while - absolutely great! Especially since i dont need a USB cable across my room since i use it with a pi over network!

tatarize, 2023-08-07:
Well, there's other devices, like if you run an ezcad2 fiber laser it's not that much more than the rest of the setup and controlled with the regular board. And it has it's own sorts of commands and doesn't replace an axis. Some other grbl devices equally can let you run the rotary without taking up an axis. Then you have x, y, and rotary axis each is independent and there's some paradigm shifts which makes that setup very unlike other setups.

It gets progressively more difficult when thinking about more and more potential setups which sort of lead to decision paralysis and thus the decision to just hack in the axis replacing rotaries at a minimum and deal with the others issues later.

tatarize, 2023-08-07:
Weirder with some table rotaries where you basically use a chuck-style rotary to cycle in new objects. Like a carousel of pens to get marked. And set up a sort of program setup situation to make that easier. In that case the rotary command itself doesn't change anything about any axis.

StefanX13, 2023-08-07:
I see. Fibe laser would be great, but a little bit out of my hobby budget ;) But i clearly see your point.

tatarize, 2023-08-07:
And there's even crazy things like rotaries on the outside of a ring to fire lasers inside the ring that is held on the outside by the chuck but at a bit of an angle which means the resulting split area inside the ring is not only a bit weird but it's also sort of a trapezoidal section. --- The more I think about it the more I realize there isn't a one size fits all solution.

StefanX13, 2023-08-07:
I am watching a video about the ComMarke B4 at the moment which shows exactly this ;)

tatarize, 2023-08-07:
Okay. I pushed 0.9.0006 (0.9.1 Beta 5) and it should correct this issue.

StefanX13, 2023-08-07:
At the moment im using 0.8.700 on my windows pc and 0.8.600 on my rpi for the network. Is its save to switch to 0.9.0006 on windows, and even more important can i mix the 2 versions, using 0.9.x on windows and 0.8.xxx on the rpi?

tatarize, 2023-08-07:
Should generally be safe, but I'm planning on updating 0.8.x too in short order. The version switch didn't seem break much if anything but the drop of features in 0.8.x means it'll probably get more stable as time go on since it just is getting bug fixes.

StefanX13, 2023-08-07:
Can confirm that it works in 0.9.0006 now. Great! Thanks a lot for this fast work!
Last question, can i use the the new features in 0.9.xxx when the network version on the rpi is still 0.8.xxx, or are the features depending on the version which is connected to the k40?

tatarize, 2023-08-07:
https://github.com/meerk40t/meerk40t/releases/tag/0.8.8000

Has the correction too. Since it's only dealing with bug fixes and no new features will get added to the 0.8.x series it should be fine.

tatarize, 2023-08-07:
The network protocol for the M2 Nano hasn't changed in a very long time. It should actually work with massive version differences. Like 0.7.x stuff for the raspberry pi shouldn't have an issues. It's literally just sending raw characters of the data that the laser should send. And could almost be stand-alone for that stuff.

StefanX13, 2023-08-07:
Good to know! Then i will use the legacy for "production" but will have closer look on the new 0.9x Version and the new features!

tatarize, 2023-08-07:
Yeah, the newer version stuff in 0.9.x wouldn't throw any wrenches into anything and the features wouldn't get in the way, but also there's no changes to the underlying protocol until that version gets out of beta, I need to avoid anything that could possibly send things into version hell and require a lot of work to fix. So I not only avoided breaking anything, but I'm very careful to avoid anything that might cause serious issues or big changes.

Basically the big shift is moving to a different underlying geometric path structure, and a builds a bunch of scaffolding to make sure that works. It broke a couple things where you needed to alter things within the underlying geometry (for example node-edit), but I don't want to end up spending many months without the new version. So I wouldn't mess with the data the drivers write, since that's asking for trouble.

StefanX13, 2023-08-07:
Can confirm that it works in 0.8.8 now.
Btw. should i open a new bug for the problem with the window settings, e.g. Laser control panel a.s.o.? They also doesnt save or load, but i cant find the right settings in the cfg

tatarize, 2023-08-07:
Most of those don't save. The pane it's on etc.

StefanX13, 2023-08-07:
A, ok then its work as intended. Thanks four your effort again! Since its 00:38 ill have a bit of sleep now ;)

closed 2023-08-07 by StefanX13

tatarize, 2023-08-07:
There's a power/speed setting for grbl that is runtime. Optimize settings seem to save fine. Move Laser should save some but I forget the specifics. If there's something else you want or thing should save... yeah raise an issue.