meerk40t/meerk40t #2042: [0.8.8] Minor bug: Spooler history start date format is not localised

Sophist-UK, 2023-08-11:
Regardless of my Windows system's date format settings (which on my system is `dd/mm/yyyy`) the spooler history start date is shown in US `mm/dd/yyyy` format.

[image]

Date needs to be shown according to the O/S locale setting.

jpirnay, 2023-08-14:
@tatarize wxMeerk40t.py states:
```python
    def InitLocale(self):
        import sys

        if sys.platform.startswith("win") and sys.version_info > (3, 8):
            import locale

            locale.setlocale(locale.LC_ALL, "C")
```
which imho should read:
```
            locale.setlocale(locale.LC_ALL, "")
```
That would fix the issue, but I remember vaguely some  comments about a locale bug?!

tatarize, 2023-08-14:
The Locale bug was a terrible bug that antagonized us for years. That was the first fix for it. Later versions of WX after 4.0.7 should have it fixed as well. But, 4.0.7 with python 3.7 would crash for people on startup if their locale was set to a sort of sub-locale which made the bug nearly impossible to recreate locally and for a while we produced a second version of the exe just to avoid that bug.

#242 was a giant pain the ass.

It's only true for wxPython 3.8+ and wx4.0.7 was pretty safe. So we locked version at 4.0.7 3.7 for a while there. It should be fixed after wx4.1.2+ meaning wx4.2.0+ too. And it only showed up fairly intermittently so its impossible to test.

tatarize, 2023-08-14:
https://github.com/wxWidgets/Phoenix/commit/040c59fd991cd08174b5acee7de9418c23c9de33

Basically Python 3.8 set the local to the local locale and the wxWidgets expected to have freedom there but didn't and crashed. I know this bug is fixable but there's some chance that this could, for some installs and source versions actually cause the bug again, and thus sort of scares me.

tatarize, 2023-08-15:
For what it's worth, I sort of prefer YYYY/MM/DD as it's generally better for sorting and a bit more international. Also, it doesn't really matter, the data is stored in a fractional unixtime value + duration.

closed 2023-08-26 by tatarize