meerk40t/meerk40t #2048: [0.8.8] (PNG) Image not loading

Sophist-UK, 2023-08-16:
The image was originally 7.5MB and I thought it was an issue with being too big, but when I scaled it to 50% size (2MB) and it still didn't load, it seems to be something else.

I will send @tatarize the SVG file by email as the image is a customer copyright, so I can't post it here.

Sophist-UK, 2023-08-16:
I saved the image as a JPG and tried using that and it still didn't load. I also started MK from a command line in the hope that there might be messages to the cmd.exe window, but nothing.

tatarize, 2023-08-16:
My inkscape which is a bit old sent me:
** (org.inkscape.Inkscape:9004): WARNING **: 11:13:11.145: shape reference on non-text object: 7SPGroup

As warnings 8 times. But the image itself loaded.

So far no issue. Lemme update my inkscape.

tatarize, 2023-08-16:
Inkscape 1.3 image is missing.

tatarize, 2023-08-16:
The issue appears to be internal to meek40t since the temp.svg file created by inkscape is actually fine and contains the image but the image does not load.

Sophist-UK, 2023-08-16:
Yes - but I tried all the techniques I have previously used to get images to load i.e. switch from png to jpg, reduce the size and nothing works. The image I used has been used in designs before, and these still load. I reprocessed it with paint,net and imported it into the SVG file.

I can load the exact same image directly into MK just fine, just not when it is embedded in the SVG file.

jpirnay, 2023-08-16:
```
python
            # Inkscape 1.3 puts some lf (\n) in the data string for
            # readability, this causes an error in svgelements, so lets
            #  fix this until svgelements get fixed
            if element.url is not None:
                element.url = element.url.replace("\n", " ")
                match = REGEX_DATA_URL.match(element.url)
                if match:
                    # Data URL
                    element.media_type = match.group(1).split(";")
                    element.data = match.group(2)
                    if "base64" in element.media_type:
                        from base64 import b64decode
                        element.data = b64decode(element.data)
                    else:
                        from urllib.parse import unquote_to_bytes
                        element.data = unquote_to_bytes(element.data)
```
We might need new versions...

Sophist-UK, 2023-08-16:
I have a feeling we had a discussion about that before - because Inkscape (1.0?) was doing this stuff and when we saved an SVG we weren't. Not a bug, just doing it differently.

BUT the bigger issue IMO is that MK was not reporting a failed image load (in this case due to what MK thinks is a malformed SVG file).

(When I was helping with code, there were numerous examples in the code where errors were deliberately being ignored rather than hunting down the cause. If you want reliable software, make sure that the code is strongly typed and don't deliberately ignore errors e.g. where the wrong arguments are being passed.)

tatarize, 2023-08-16:
I think the issue might be Pillow. I didn't fully check it out. But, the error it said was that the image was Truncated or something giving an OSError.

tatarize, 2023-08-16:
Wait no. Double checked the data says:

b'\x89PNG\r\n\x1a\n\x00\x00\x00\rIHDR\x00\x00\x056\x00\x00\x06\x84\x08\x06\x00\x00\x00\xc9\xc0A\xdb\x00\x00\x01iiCCPICC Profile\x00\x00(\xcf}'

That's not pillow but svgelements failing to correctly extract the bytes I guess.

tatarize, 2023-08-16:
Okay, and I finally caught up the REGEX_DATA_URL match in svgelements does not process the `\n` line ends when it processes the given self.url.

It's not that common to match multi-line bits of code. 

`REGEX_DATA_URL = re.compile(r"^data:([^,]*),(.*)", re.MULTILINE)` (doesn't seem like it fixes it)

For the regex might work, also removing the lineends and a few more things.

I'll figure it out and start pushing the data around since this is a big pain in the butt, since it requires a fix to svgelements etc. (meaning writing tests and pushing out a new version etc)

tatarize, 2023-08-16:
match = REGEX_DATA_URL.match(self.url)

match = REGEX_DATA_URL.match(self.url.replace("\n", ""))

at line 8530 might be properly successful.

tatarize, 2023-08-17:
Confirmed fixed. #2045

This is flagged for 0.8.8 so I have very little qualms bumping the version up and releasing a new version since we're dealing with a bug fix applied to a legacy bug-fix branch branch.

closed 2023-08-26 by tatarize