meerk40t/meerk40t #2050: [Suggestion]:

LIBERTY-UK, 2023-08-17:
### Summary Description

Control of trace (frame) speed in Lightburn

### Current functionality

I use MeerK40t in Ruidacontrol mode so I can use K40 laser with Lightburn.  In 'normal' lightburn use, the frame function to trace the outline before committing to cut, can have its speed altered.  When running through MeerK40t however, the facility to adjust the frame speed and indeed the move speed is inoperative.  This means the frame speed is essentially going at high speed.   Likewise the 'move' speed is also inoperative when using through MeerK40t.

### Future functionality

A separate self contained 'utility program' for those that just want the ruida emulator.

### MeerK40t Version

8

### MeerK40t Type

Source (zip/tar file from Github)

### Your Operating System

Windows

tatarize, 2023-08-18:
The frame speed does not exist within MeerK40t. Things typically frame at whatever speed they are set to frame at. For the most-part. I think some of them like GRBL might require a speed and picks one but its certainly not standardized. Lightburn however sends this frame as a legit move to these points at this speed operation. So that happens. Per standard, meerk40t isn't doing anything other than translating the formats and trying to figure out which rasters can be done at a rastering speed.

As for future functionality, you could actually run ruidacontrol from console. `meerk40t -Zce ruidacontrol` would, in theory work.  It wouldn't load up the gui and would run ruidacontrol. Though this might be better done with `grblcontrol` running on a raspberry pi (ruida signals are sent through lossy UDP and I don't trust them through the actual network).

LIBERTY-UK, 2023-08-18:
Thanks for the detailed reply.  At the moment I use K40 Whisperer, which as you are aware, has the facility to define the trace speed. The adjustable trace speed allows users to ensure the substrate is in **exactly** the right place, before committing to cut (or engrave).  Lightburn trace (frame) just going too fast.

I will have a play around with running Lightburn from console.  All about trying new directions.. Thanks..

tatarize, 2023-08-18:
You mean meerk40t? Lightburn's console stuff is pretty bad

LIBERTY-UK, 2023-08-18:
Yep I did mean MeerK40T..  Personally I think Lightburn is way over rated and over the top for most people's needs.  Certainly wouldn't invest in an expensive board conversion to run it on a K40 machine.  I have spare boards so staying the course with them.  

Be nice to think that MeerK40T might bridge that gap at some point.  For my own personal needs though, it is currently the lack of adjustable trace speed that is the killer.

Sophist-UK, 2023-08-18:
Actually I think that there are a bunch of potential enhancements to MK in the area of tracing the outline other than just speed.

For example, turning on the laser at very low power whilst doing a rectangular or hull outline, automatically burning dots at the corners of a rectangular outline, etc.

tatarize, 2023-08-19:
Yeah, I think Sophist is right and you are right as well. There's clearly a need for adjustable trace speed. I say this having owned a GRBL for about a day. When it decides to go way too fast it's really really annoying.

tatarize, 2023-08-19:
Also, we have a bunch of other would-be traces we can do. And there's a sort of standing gap in trace with regards to placements. In effect if you set a grid of placements and you tell it to trace you should actually get the entire grid. As is, it wouldn't even trace a current position flagged placement correctly.