meerk40t/meerk40t #2069: [0.8.9] Spool history - Put file name in Jobinfo column.

Sophist-UK, 2023-08-26:


jpirnay, 2023-08-26:
[image]

That's already there - or do you refer to the complete filename including path?

tatarize, 2023-08-26:
And sometimes the name is a bit wishywashy since you loaded it and changed stuff in there. Which is why the name is only the filename sometimes but usually Spooler: x items

Sophist-UK, 2023-08-27:
I would like to propose the following to make things less wishy-washy.

1. There needs to be a "current" filename based on the following rules:
    1. If the last load/save action was a save, then the save filename.
    2. If the last action was a load and there is one file node and no other children of the elements super-node, the file in the one file node.
    3. If the last action was a load and rule 2 does not apply - the filename should be empty.
2. The "current" filename is the filename used for Save - and Save As would be shown if the "current" filename is empty.
3. The "current" filename is what is used for the spooler - if the filename is empty then you do as shown above. I am not sure that showing a full path is necessary, but if it is then it should probably be a preference.
4. We need to indicate whether there have been any changes to the design i.e. that a save would be different from the load - perhaps by prepending an asterisk to the menu bar title like Inkscape does. So any design or operation changes (including creating Ops or using existing Ops when loading a file which doesn't have saved Ops) would indicate a change. This state should be put on the Undo record - so if you save, then make a change and then undo it, you get the save-needed indicator cleared (just as Inkscape does).

These changes would make the save functionality a bit more intuitive - it would distinguish between MK saved SVG files (which have Ops) and those that don't and would be more intuitive about when Save As is shown if you do a Save (i.e. if I Save As then remove the file node and import another, then if I hit Save I think it will currently overwrite the previous save even though the file contents are entirely different).

tatarize, 2023-08-29:
Working file name exists, this is common for save. I think it displays in taskbar that's what's believed to be the current file. But, you can load multiple files to build a single project.

3. This is somewhat an issue, sometimes jog commands etc are sent to the spooler and those aren't actually part of the filename as such.  Saying that a jog command was that file would be wrong.

4. This seems plenty doable, I think we have a flag for changed data or whatever. Though I think suggesting this would make it all more intuitive seems to be overselling that.

If we wanted to better distinguish MK and SVG files we could probably just make a different extension, part of the issue there is while meerk40t is adding more and more info there, it's still at the core a proper SVG.

closed 2024-10-04 by jpirnay