meerk40t/meerk40t #2072: [0.8.9] Weird extra job spooled

Sophist-UK, 2023-08-27:
I just hit the BGB and got this in the spooler:
[image]
I have no idea what the second job is.

tatarize, 2023-08-27:
I assume BGB is big go button... Yeah, that's rightly weird.

Sophist-UK, 2023-08-27:
Big Green Button.

I may have double-clicked it. The second job was a duplicate. So the primary issue is that double-click = double job. Secondary issue is why the steps were wrong.

tatarize, 2023-08-27:
That happens. I think it's actually because you may have double clicked a thing and or triggered it inadvertently that it's sketchy with double clicking. It may actually just be there to prevent you from double clicking like that.

tatarize, 2023-08-27:
I think this relates to #1870 to only allow conscious clicks. Still spools it but disables that because your click was invalid or something.

tatarize, 2023-08-27:
Nope. Didn't pan out. Maybe `plan clear` is breaking the LaserJob already generated?

Yeah, you can click the BGB and get it to happen. So it's quite minor, and easy to reproduce.

jpirnay, 2023-08-27:
Double Click should be suppressed now.
The steps will only be updates, when the job starts.

closed 2023-10-19 by tatarize