meerk40t/meerk40t #2073: [0.8.9] Cancel issues

Sophist-UK, 2023-08-27:
m2 nano...

When cancelling during a cut...

1. Cancel does not stop immediately - m2 nano continues cutting for more than the currently straight-line segment, but **NOT** the entire buffer to the point shown by the reticule on screen.
2. It then appears to do a return to origin, but from the point at the end of the buffer - this results in a graunching of the stepper motor as it hist the end stops.
3. (Probably not fixable) Occassionally USB stops working altogether and **only** solution is a power recycle of the m2 nano - replugging the USB, or playing with MK controller menu/buttons do not work.

If MK is indeed attempting a move to origin, this should IMO be removed. Instead, the driver should note that position has been lost and it should either:

1. Insert a home at the start of the next job (if user doesn't already have an auto-home set as the first job-start default action); or
2. At the start of the next job, warn the user that position is lost and that either a home is required or current position should be assumed to be origin. (Even better, also have a Device option to ask, home or assume origin.)

Lost position should also be assumed when:
1. MK is started
2. USB is connected when there is no job being processed.

jpirnay, 2023-08-27:
As far as I can see there is no logic to move to home / physical home after an estop. And I haven't witnessed such a thing either. So it might be some residual ops/buffer? For tat to answer

Sophist-UK, 2023-08-27:
It is reproducible.

closed 2024-10-04 by jpirnay