meerk40t/meerk40t #2075: [0.8.9] Crash in tree control

Sophist-UK, 2023-08-27:
I was doing something else at the time this happened. I may have nudged the mouse or keyboard, but no idea what.
```
MeerK40t crash log. Version: 0.8.9000 on Windows: Python 3.8.10: AMD64 - wxPython: 4.2.1 msw (phoenix) wxWidgets 3.2.2.1
Traceback (most recent call last):
  File "meerk40t\gui\wxmtree.py", line 1609, in on_drag_end_handler
  File "meerk40t\core\elements\elements.py", line 1552, in drag_and_drop
  File "meerk40t\core\node\groupnode.py", line 62, in drop
  File "meerk40t\core\node\node.py", line 1043, in append_child
  File "meerk40t\core\node\node.py", line 641, in notify_attached
  File "meerk40t\core\node\rootnode.py", line 55, in notify_attached
  File "meerk40t\gui\wxmtree.py", line 482, in node_attached
  File "meerk40t\gui\wxmtree.py", line 1066, in node_register
wx._core.wxAssertionError: C++ assertion ""param"" failed at ..\..\src\msw\treectrl.cpp(1113) in wxTreeCtrl::SetItemData(): failed to change tree items data
```

Sophist-UK, 2023-08-27:
Looks like it was some sort of delete which I was able to undo.

tatarize, 2023-08-29:
This may relate to another set of crashes that I cannot ever track down. Like there's no repeatability or understanding how that unregistered node got in there or whatever.

I can't solve these bugs. It's annoying but I've seen similar crash logs many times and they've never been solved. If you ever figure out a method for consistently recreating this, it would be awesome.

closed 2024-10-04 by jpirnay