meerk40t/meerk40t #2076: [0.9.0rc1] Issues

Sophist-UK, 2023-08-27:
The first RELEASE CANDIDATE release of 0.9.0 is Release Candidate 1 (0.9.0010). This issue is for reporting issues with this specific version of MeerK40t.

Please do try this release, although even as a Release Candidate there is no guarantee that this is suitable for production use. Please do not report any further bugs with previous beta releases or indeed any releases other than 0.9.0 RC 1.

This issue report wants to cover identified bugs / issues with this specific version of Meerk40t to help resolve them. Please be specific with your report and if feasible attach relevant information (screenshots, the file you used etc.) to it, so that the developer team can try to recreate the issue.

If you want to provide feedback on other subjects close to your heart please use appropriate channels:

General discussion: https://github.com/meerk40t/meerk40t/discussions
Feature requests: Please open a separate issue in this channel using [THIS LINK](https://github.com/meerk40t/meerk40t/issues/new?assignees=tatarize&labels=Type%3A+Enhancement&template=suggestion.yml&title=%5BSuggestion%5D%3A).

Sophist-UK, 2023-08-27:
I will certainly be testing out this release candidate. I am not sure whether the potential issues discussed in #2022 have been resolved, but IMO in order that the full 0.9.1 release looks polished these do ideally need to be resolved before it is published.

Sophist-UK, 2023-08-27:
Extremely minor point - IMO the Language menu should either:

a. Have all languages translated into the currently selected language i.e. for default English language, all listed languages should be in Latin script (however I accept that it probably makes more sense to have languages listed in their own native script/language); or

b. Have Latin script languages first in alphabetical order (case insensitive), then followed by non-latin languages in some sensible order (no idea what that might be because I do not really understand any non-latin languages).

Sophist-UK, 2023-08-27:
Even more minor point: the tabs for the new ribbon bar do not have sufficient bottom margin and descender characters (such as lower-case "g" in "Config" actually cross the border of the boxes of the ribbon bar.

Sophist-UK, 2023-08-27:
The Tools pane is still not formatting well. 

When used in a horizontal panel, I would say they are formatting and scaling very well.

However, (in addition to my comments in #2022) when used in a vertical panel:,

1. Labels are being cut off and even narrowing the panel to make the icons smaller still results in some buttons having their labels cut off. I am not sure what the solution is for this as making the icons or text smaller is not a great solution, and reducing the outer margins (outside the group boxes) will only help a tiny bit. But it looks ugly and unfriendly. (Perhaps labels should simply not be shown.)
2. Widening the vertical panel does **NOT** result in the icon groups being displayed with multiple icons horizontally i.e. the first 14 icons could be grouped in a 2x7 array allowing more space for each row of icons, but they remain at 1 icon per row. IMO this is the fix that is needed for point 1.
[image]
3. Crazily, when the panel width reaches 51%, despite it **still** being a vertical panel, the tools bar switches to a horizontal layout with full height icon boxes (which looks just bonkers). IMO a vertical panel should remain a vertical panel regardless of its width i.e. the icon **groups** should stay stacked vertically even if there is enough width to place them horizontally).
[image]
4. The lack of colour cues in all the panes is now becoming a significant negative IMO - the panes are now essentially monochrome. I really cannot put my finger on why this feels quite so bad for me (except to say that perhaps the greyness is depressing), but for me at least I really feel that I need some colour cues despite being familiar with MK and this might be more important for newcomers. I should add I have a similar problem with recent versions of Inkscape which has by default a monochrome theme, but Inkscapes monochrome icons are **much** better quality **and** Inkscape allows me easily to change to a colour scheme. 
   
    (As a humurous aside, I thought that black & white monitors had become a thing of the past several decades ago. Even in the late 1980s colour monitors for mainframes were becoming common, and PCs probably stopped having monochrome monitors at around the same time. Certainly I cannot remember **ever** seeing a Windows 3.11 PC with a monochrome monitor and that was early 1990s.) 

5. The icons are low-res / grainy which makes them difficult to understand when you are focused on them and searching for something that will do what you want, and when you know what you are seeking the one you want does **not** immediately jump out at you. Some are too detailed, on some the lines are too thin, on some they don't have anti-aliasing. But a lack of colour also doesn't help. And the net result is that this is a significant issue IMO. As an easy fix you might want to think about copying Inkscape's icons - Inkscape is GPL so you may be able to use them without any issues.

I still need to try this RC1 candidate on my production PC, and I am not certain whether these will be significant enough to prevent me using this in production.

tatarize, 2023-09-04:
The language thing is easier since we don't need to translate them, but also trying to find the Chinese script that says English isn't fun and I've had to do that a few times with other programs.

tatarize, 2023-09-04:
My descender looked to have enough room.

[image]

Sophist-UK, 2023-09-04:
Actually, it looks OK on my laptop too.
[image]
I will have to check on the Surface Pro 3 I use for production.

Sophist-UK, 2023-09-04:
Tools pane in a vertical panel no longer seems to go horizontal when it goes above 50% width. 

But labels are still partially hidden, tools icons are still fuzzy and it still won't go to e.g. 2 icons per row when width allows it.

closed 2024-02-19 by jpirnay