meerk40t/meerk40t #2078: Anonfiles shutdown crash logs no longer work.

tatarize, 2023-08-28:
This will probably require that we move to a proper domain. Most likely this will be https://meerk40t.com which probably would be running drupal version 9.

In theory this would be best if the server did some cursory scanning of the crash log and posted it via webhooks to the discord. But, for now I'll have my brother set a place for these files to go.

Sadly the writing was on the wall but I kept kicking the can further down the road, and while ancient crash logs aren't that helpful the newer ones are a pretty vital needed thing.

Last crash logs were processed and corrected 7/31/23.

Sophist-UK, 2023-08-28:
Why drupal rather than (say) Laravel?

I would suggest that your starting point would be to list the totality of functionality you would want for this site. My own suggestions would be: Static "advertising" pages, documentation (using a wiki?), a blog, and some bespoke functionality for e.g. crash logs etc.

IMO Laravel (plus some extensions like Folio for static pages, Bookstack for a wiki and one of a number of extensions for a blog) would be the best solution.

tatarize, 2023-08-29:
My brother already has some Drupal servers, and the hostname. The main thing would be to just set the endpoint up to receive that content. Perhaps even returning a reply like "that bug was already fixed in the latest version" or such. Basically the files aren't going anywhere, and I sort of need those logs. At first when @inspectionsbybob suggested it, it was a bit meh... but now it's sort of critical to have that information.

Sophist-UK, 2023-08-29:
I am not suggesting that the crash file functionality isn't valuable, just that with 0.9 imminent this functionality could be re-enabled in 0.9 and the impact of it not being in 0.8 is going to be significantly smaller and perhaps simply not enough to delay the next 0.8 full release. (Or you could put out a 0.8 RC so the rest of the bug fixes get out there.)

As for Drupal, obviously you need the crash functionality fast, and if your brother has the Drupal experience and part build complete, using this as a stop-gap is a good idea. But IMO you really need to think strategically about what you need the site to deliver, and Drupal probably isn't the best technology for this these days.

You also know my views about using the community to boost your resources - and this (i.e. documentation) is perhaps the prime example of where IMO you will make faster progress with better quality by engaging with the community to achieve a strategic solution rather than making a tactical decision.

tatarize, 2023-08-31:
Most 0.8.x bugs are going to affect 0.9.x too. Mostly all that's being done for 0.8.x going forward are bug fixes. And disabling some things that never worked very well.

Clearly older version of the 0.8.x line can't be fixed but later versions certainly could.

My brother suggested just setting up a basic flask api in python since a full implementation of anything there is probably going to overkill. We have an old blade server setup so it's not that weird to make it sit there and gather some crash logs. And it would probably give me some fine grain control to decide on a reply. So submitting a crash report could trigger a few more things fairly easily. Namely we could send proper secured commands to do a discord webhook, or a maybe post something to github (since the endpoint could have proper tokens to open an issue). Though I'd want to check through some of that, since you could in theory send any random file to the given endpoint.

It could also maybe analyze the report and reply to the post with a response such as, "this crash was previously reported and was fixed in the latest version."

Sophist-UK, 2023-08-31:
If all you want is to process appcrash reports, any working technology should do. In which case I would advise setting up a `crashreports.meerk40t.com` subdomain to host it on. Indeed, this might be a sensible separation of functions anyway.

As I have said elsewhere, if you just want to write code and are not interested in getting Meerk40t out to the widest user base, you don't need documentation, and if you are not bothered about getting a wide range of user input on UX, nor getting assistance with producing documentation (because your own time is finite and you want to focus on the code), then you don't need a decent web site with a wiki etc. BUT...

If you want to harness the enthusiasm of the user community to give you UX input and to create documentation as a way of giving back for the free use of your great software then you probably need to invest in giving the community the tools to do that. (You can probably even get the community to build the site for you if you are willing to be open to that. You don't have to do everything yourself.)

tatarize, 2023-09-01:
This stuff got put under http://dev.meerk40t.com and should work fine. I added in a basic ability to submit files and some pre-ordained dummy files for `bug` and `feature_request` commands. This stuff also makes a call to the discord webhooks and notifies `#dev`. It's decidedly not fancy, but it's fancier than I had before.

There's certainly some work to be done with checking diffs of crash reports and determining crash clusters.

There's also a small but generally unused mechanism that allows the file submission to reply with a message, in theory this would allow, if I have the crash cluster processing done right, to determine if this crash submission was a fixed bug and notify the user. `"This crash was submitted, however it was fixed in version 0.9.4, and will not be addressed."` -- which is simply made possible but probably won't be implemented currently it just accepts crash logs and sends that data to discord via a webhook. The bug and feature_request commands also just send a file though one with a specific name. And any future functionality there would also just be sending a file in that same manner. --- Or will always just have that as a backwards compatibility thing.

closed 2023-09-13 by tatarize