meerk40t/meerk40t #2079: [Suggestion] Special op to split a job into separate spool jobs

Sophist-UK, 2023-08-28:


tatarize, 2023-08-28:
Hm. What would be the use-case for that? I mean it won't optimize across anything there and mostly the job is spooled independently as a single object. But, yeah, it could be divided into different sections.

tatarize, 2023-08-28:
And would this sort of thing happen often enough that you wouldn't be better off spooling the first few operations and then the second few operations?

tatarize, 2023-08-29:
Also, much like how lightjobs works it could be set to run a job forever. And let the user stop the job, or cancel that part of the job and move on to the next part of the job.

The laserjob are really just a list of plotplanned data. They are passed the driver for the laser and proceed to send that data. But, things like the infinite loops (You'll fine it in the properties of the Operation branch itself) just have that send the data forever. It seems a bit like, if the goal here is to run until done there's other ways to do that. Probably by turning some operations off, running an infinite job, then toggling for the other operations.

jpirnay, 2023-10-24:
We will have an opportunity to simply add additional loops to a running job and the opportunity as well to stop a job after the completion of the current pass, that may already cover most of the use cases outlined here

closed 2024-10-04 by jpirnay